ceccocats/tkDNN #3: Fails while exporting yolov4

Kmarconi, 2020-04-29:
Hi ! FIrst thanks for your repo, I saw what you've done with the new yolov4 weights in terms of FPS and I wanted to reproduce it. I'm working on a Jetson Xavier and when I'm trying to export the yolov4 weight, I got this error at for the 107 layer : darknet: ./src/darknet.c:519: run_export: Assertion `0 == "layer type not supported for export"' failed. Do you know why this error is showing up ?

Kmarconi, 2020-04-29:
By commenting out the 519 line of the darknet.c file, i'm able to export my weights, don't know if t's a good idea

ceccocats, 2020-04-29:
Hi,
The export in our darknet fork was wrong (it missed a layer in the export function).
We fixed it and now should work.
Please check you are using the correct repo:
https://git.hipert.unimore.it/fgatti/darknet

Anyway if you need to test yolo4 with the official weights you dont need to export.
You just run the ./test_yolo4 sample and it will automatically download it.

Kmarconi, 2020-04-30:
Thanks for your reply ! :) Yeah i saw yesterday that by doing directly ./test_yolo4 it is autommaticaly downloading it :) Just so you know, I'm arround 34 FPS on my Xavier with yolov4

closed 2020-06-01 by ceccocats