ceccocats/tkDNN #6: Add support to YoloV3-Tiny-PRN

marcoslucianops, 2020-05-03:
Can you add support to [YoloV3-Tiny-PRN](https://github.com/WongKinYiu/PartialResidualNetworks)?

[image]

ceccocats, 2020-05-04:
Hi,
Although this net is very small and easy to port we cant implement every network in the world.
In this repo there are the dnn that are interesting for us.
Since we use a lot of work from darknet in the future I probably will create a parser for darknet cfg to automatically import; But right now you have to manually write your own network.

ceccocats, 2020-06-01:
The parser is now avaible
https://github.com/ceccocats/tkDNN#darknet-parser

closed 2020-06-25 by ceccocats