ceccocats/tkDNN #7: Get output of each layers

CiRA-AMI, 2020-05-06:
Hello
Thank you for your work. It's amazing.
I want to construct feature map picture.
May I ask you about how to get output of each layers.

ceccocats, 2020-05-07:
Hi,
If you have a _tk::dnn::Network_ object you can iterate the Layers pointers:
```
for(int i=0; i<net.num_layers; i++) {
    tk::dnn::Layer *l = net.layers[i];
    l->dstData  // pointer to output data 
    l->output_dim // output data dimension 
}
```
NB: dstData is a CUDA memory pointer so you can't directly access it from CPU
You can also keep the layer pointer while you are adding layers to the network.

If you use _tk::dnn::NetworkRT_ is not possible without writing a proper TensorRT plugin.

Anyway I would like to see the results :)

AleBasso80, 2020-05-12:
Hello,
if I have a Yolov4 how can I do that?
Thanks!

ceccocats, 2020-05-12:
If you use the test: https://github.com/ceccocats/tkDNN/blob/master/tests/yolo4/yolo4.cpp

AleBasso80, 2020-05-13:
Hello @ceccocats,
first of all thank you for your reply. 
Unfortunately in normal operation I don't have enough time to build the RT network at each system start, therefore I need to get it loaded from file. As a consequence, I think that the network layer pointers are not available, therefore I can't follow your suggestion. I was wondering if I can achieve the goal by modyfing the PluginFactory class in NetworkRT.h file, so that it returns, in addition to the yolo layer pointers, a pointer to some other layers I'm interested in. Then I get the data as you get the yolo's. Do you think this may be the right approach?
Thank you.

ceccocats, 2020-05-13:
You can make a fake PlugIn the copy the output to some memory.
But there is anothe simplier option, you cat set another layer as the output of the network, for example:
```
     c160.setFinal();
```
this will set the c160 as an additional output of the network so the tensorRT model will have now 4 outputs (3 yolo + c160).

NB: probably this edit will broke something in the result check and in the post process of the net results.
Because I expect only 3 outputs and I will get 4, for example in test_yolo4 (line 648):
```
    for (int i = 0; i < 3; i++)
        rt_out[i] = (dnnType *)netRT.buffersRT[i + 1];
```
In this code I take the pointers of the 3 outputs. buffersRT contains [ input, output_yolo0, output_yolo1, output_yolo2]
Depending on the order of the layers your output will be in that array

AleBasso80, 2020-05-13:
Thank you for the hint. I'll explore both ideas.
Thanks!

AleBasso80, 2020-05-13:
@ceccocats I don't understand where I must set the setFinal() options.

closed 2020-06-25 by ceccocats