ceccocats/tkDNN #8: What can I automatic create tk::dnn struct file?

georgetseng1128, 2020-05-08:
Hi, we have run example of yolov3 and yolov3-tiny.

We will try self darknet model, first export bin file by:
`./darknet export xxx.cfg xxx.weight layers`

But see example of yolov3, the yolov3 tk:dnn struct is in include/tkDNN/models/Yolo3.h

So, if new model, we will manual create new model tk:dnn struct?
Do you have supported automatic create tk:dnn struct file?

About include/tkDNN/models/Yolo3.h:
```
int preYoloFilters = (classes+5)*3; 

std::string input_bin = bin_path + "/layers/input.bin";
std::vector<std::string> output_bins = {
    bin_path + "/debug/layer82_out.bin",
    bin_path + "/debug/layer94_out.bin",
    bin_path + "/debug/layer106_out.bin"
};
std::string c0_bin    = bin_path + "/layers/c0.bin";
std::string c1_bin    = bin_path + "/layers/c1.bin";
std::string c2_bin    = bin_path + "/layers/c2.bin";
std::string c3_bin    = bin_path + "/layers/c3.bin";
std::string c5_bin    = bin_path + "/layers/c5.bin";
std::string c6_bin    = bin_path + "/layers/c6.bin";
std::string c7_bin    = bin_path + "/layers/c7.bin";
std::string c9_bin    = bin_path + "/layers/c9.bin";
std::string c10_bin   = bin_path + "/layers/c10.bin";
std::string c12_bin   = bin_path + "/layers/c12.bin";
std::string c13_bin   = bin_path + "/layers/c13.bin";
std::string c14_bin   = bin_path + "/layers/c14.bin";
std::string c16_bin   = bin_path + "/layers/c16.bin";
std::string c17_bin   = bin_path + "/layers/c17.bin";
std::string c19_bin   = bin_path + "/layers/c19.bin";
std::string c20_bin   = bin_path + "/layers/c20.bin";
std::string c22_bin   = bin_path + "/layers/c22.bin";
std::string c23_bin   = bin_path + "/layers/c23.bin";
std::string c25_bin   = bin_path + "/layers/c25.bin";
std::string c26_bin   = bin_path + "/layers/c26.bin";
std::string c28_bin   = bin_path + "/layers/c28.bin";
std::string c29_bin   = bin_path + "/layers/c29.bin";
std::string c31_bin   = bin_path + "/layers/c31.bin";
std::string c32_bin   = bin_path + "/layers/c32.bin";
std::string c34_bin   = bin_path + "/layers/c34.bin";
std::string c35_bin   = bin_path + "/layers/c35.bin";
std::string c37_bin   = bin_path + "/layers/c37.bin";
std::string c38_bin   = bin_path + "/layers/c38.bin";
std::string c39_bin   = bin_path + "/layers/c39.bin";
std::string c41_bin   = bin_path + "/layers/c41.bin";
std::string c42_bin   = bin_path + "/layers/c42.bin";
std::string c44_bin   = bin_path + "/layers/c44.bin";
std::string c45_bin   = bin_path + "/layers/c45.bin";
std::string c47_bin   = bin_path + "/layers/c47.bin";
std::string c48_bin   = bin_path + "/layers/c48.bin";
std::string c50_bin   = bin_path + "/layers/c50.bin";
std::string c51_bin   = bin_path + "/layers/c51.bin";
std::string c53_bin   = bin_path + "/layers/c53.bin";
std::string c54_bin   = bin_path + "/layers/c54.bin";
std::string c56_bin   = bin_path + "/layers/c56.bin";
std::string c57_bin   = bin_path + "/layers/c57.bin";
std::string c59_bin   = bin_path + "/layers/c59.bin";
std::string c60_bin   = bin_path + "/layers/c60.bin";
std::string c62_bin   = bin_path + "/layers/c62.bin";
std::string c63_bin   = bin_path + "/layers/c63.bin";
std::string c64_bin   = bin_path + "/layers/c64.bin";
std::string c66_bin   = bin_path + "/layers/c66.bin";
std::string c67_bin   = bin_path + "/layers/c67.bin";
std::string c69_bin   = bin_path + "/layers/c69.bin";
std::string c70_bin   = bin_path + "/layers/c70.bin";
[... 207 lines cut ...]
tk::dnn::Conv2d     c96  (&net, 128, 1, 1, 1, 1, 0, 0, c96_bin, true);
tk::dnn::Activation a96  (&net, tk::dnn::ACTIVATION_LEAKY);
tk::dnn::Upsample   u97  (&net, 2);

tk::dnn::Layer *m98_layers[2] = { &u97, &s36 };
tk::dnn::Route      m98  (&net, m98_layers, 2);
tk::dnn::Conv2d     c99  (&net, 128, 1, 1, 1, 1, 0, 0, c99_bin, true);
tk::dnn::Activation a99  (&net, tk::dnn::ACTIVATION_LEAKY);
tk::dnn::Conv2d     c100 (&net, 256, 3, 3, 1, 1, 1, 1, c100_bin, true);
tk::dnn::Activation a100 (&net, tk::dnn::ACTIVATION_LEAKY);
tk::dnn::Conv2d     c101 (&net, 128, 1, 1, 1, 1, 0, 0, c101_bin, true);
tk::dnn::Activation a101 (&net, tk::dnn::ACTIVATION_LEAKY);
tk::dnn::Conv2d     c102 (&net, 256, 3, 3, 1, 1, 1, 1, c102_bin, true);
tk::dnn::Activation a102 (&net, tk::dnn::ACTIVATION_LEAKY);
tk::dnn::Conv2d     c103 (&net, 128, 1, 1, 1, 1, 0, 0, c103_bin, true);
tk::dnn::Activation a103 (&net, tk::dnn::ACTIVATION_LEAKY);

tk::dnn::Conv2d     c104 (&net, 256, 3, 3, 1, 1, 1, 1, c104_bin, true);
tk::dnn::Activation a104 (&net, tk::dnn::ACTIVATION_LEAKY);
tk::dnn::Conv2d     c105 (&net, preYoloFilters, 1, 1, 1, 1, 0, 0, c105_bin, false);
tk::dnn::Yolo      yolo2 (&net, classes, 3, g106_bin);

yolo[0] = &yolo0;
yolo[1] = &yolo1;
yolo[2] = &yolo2;

ceccocats, 2020-05-08:
Hi, that is written manually.
In the test there is a lot of ugly fast written code. The yolo3 include is not a good example.
The best thing to do is to enclose your network in a function. 
To do that you MUST alloc the layers with dynamic memory (new) because otherwise they will be deleted with the stack of the function.
A good example is this:
https://github.com/ceccocats/tkDNN/blob/master/include/tkDNN/ImuOdom.h

It is in my TODO list to write all the test in this way or to write a darknet parser; but is a lot of work. And the majority of the networks are meant to be used in tensorrt without creating every time the network in this manner.

georgetseng1128, 2020-05-13:
Okay, we look forward to your release of this way soon.
Thank.

ceccocats, 2020-06-01:
The parser is now avaible
https://github.com/ceccocats/tkDNN#darknet-parser

closed 2020-06-17 by ceccocats