ceccocats/tkDNN #10: Convert yolov4 model to tensorrt get "Wrong" error

lthbk5919, 2020-05-11:
Hi, thank for this awesome project.
I have a question about converting yolov4 model.
I try run  `test_yolo4` with yolo-608 config and get red line `Wrongs` in log as shown below:
[image]

Line `Wrong` what dose it mean? 
Thanks for the repo again.

ceccocats, 2020-05-11:
Hi,
It means that the first yolo output layer have 102 wrong values respect to the output of darknet.
We have the same errors, so I need to investigate, the strange thing is that the "correct" values from darknet are NaN.

ceccocats, 2020-05-11:
Since the wrong values are all NaN In the correct version is more a problem of darknet. The final bboxs should not be affected anyway.
The difference in result could be due to different implementation. The CUDNN version should not be to much different from darknet, the main difference is that that they uses a custom implementation of batchnorm and not the one from CUDNN.

lthbk5919, 2020-05-12:
Hi, thank you.

> The difference in result could be due to different implementation. The CUDNN version should not be to much different from darknet, the main difference is that that they uses a custom implementation of batchnorm and not the one from CUDNN

I will check it.

vasanktt, 2020-06-12:
Hi, Thank you for your great repo.

I also meet the same error while converting Yolov4 model got trained from AlexeyAB/Darknet repo. I trying to test TensorRT model that converted but it seem I didn't get the correct result but on AlexeyAB I able to get correct result.

ceccocats, 2020-06-12:
What you mean for correct result?

vasanktt, 2020-06-12:
I'm sorry for confused message. I mean I test my own Yolov4 model via `./darknet detector test .....` command and I got all correct object detected but when I export that model to TensorRT format and test it by `./demo yolo4_fp32.rt video.mp4 y` I didn't get any object.

ceccocats, 2020-06-12:
It could be due to a lot of problems
#29 is trying to do the same

vasanktt, 2020-06-15:
My issue have been solved because of I forgot to change the path of cfg file in `tests/darknet/yolo4.cpp`.
```
std::string cfg_path  = "plateocr_yolov4.cnf";
std::string name_path = "plateocr_yolov4.name";
```

@ceccocats Thank you for your support.

beizhengren, 2020-07-15:
Hi@ceccocats 

> Since the wrong values are all NaN In the correct version is more a problem of darknet. The final bboxs should not be affected anyway.
> The difference in result could be due to different [...]

By your comment,  I understand that `Nans` is the bug of darknet, and `Nans` don't affect the inference result by `./demo yolo4_fp32.rt video.mp4 y ...`.
Is that right?

mive93, 2020-07-15:
Hi @beizhengren, that is right.

beizhengren, 2020-07-17:
@mive93 Hi, thank you so much!

teplofizik, 2020-08-03:
[image]

Have such problem at jetson xavier on converting fp16/int8 mode (fp32 ok). It can run converted on another machine models (at any mode: fp32, fp16), but cannot convert models to reduced precision.

Crystalxd, 2021-03-12:
@teplofizik  Now I convert the yolo4 model to fp16, I still got this problem, but I can run the demo currently.