ceccocats/tkDNN #12: About performance of FPS with yolov3 and yolov3-tiny on Xavier

georgetseng1128, 2020-05-15:
We are interested in the performance acceleration proposed by tkDNN, so we implement tkDNN according to the workflow by you provided, and execute the examples of Yolov3, include fp32, fp16 and int8.

We have also used examples provided by NVIDIA to implement TensorRT in the past, example of Yolov3 by python like / usr / src / tensorrt / sample / python / yolov3_onnx. (we will call it TRT)

I made the FPS obtained by Darknet, TRT, tkDNN into a table, the table is follow as:
[https://cdn-images-1.medium.com/max/800/1*7SptE6sH_PXHktPttzdruA.png](url)

Expected that tkDNN will be faster than TRT,  but we observed that the speed of TRT is similar to that of tkDNN.

Why is tkDNN not faster than TRT?

ceccocats, 2020-05-15:
tkDNN is substantially a cudnn and tensorrt wrapper so is totally normal that the performance are the same.

georgetseng1128, 2020-05-18:
Hi, @ceccocats 
We asked this question because we saw this [message](https://github.com/AlexeyAB/darknet/issues/5354#issuecomment-621436453) on AlexeyAB Darknet.

In this [tabal](https://user-images.githubusercontent.com/11939259/80639851-de58a180-8a62-11ea-8479-313f0423aa57.png), It means that tkDNN(FP32/FP16) is much faster than Darknet and deepstream(TRT). So we want to know how this improves FPS?

mive93, 2020-05-18:
Hi @georgetseng1128,
I have performed those test last fall, and those were the results.
Maybe meanwhile the performances of Deepstream (ora that yolo implementation) have been improved. 
I can make some tests on the Xavier and come back to you, if you want.

georgetseng1128, 2020-05-19:
Hi @mive93 ,
Ok, please try tkdnn with yolov3 on xavier, thank.

mochechan, 2020-07-23:
Data about INT8 are still absent from cell of TX2 and NANO.
https://github.com/ceccocats/tkDNN#results

ceccocats, 2020-07-23:
That's because INT8 it is not supported by the hardware of that boards

closed 2020-09-11 by mive93