ceccocats/tkDNN #14: is Cuda 10.0 required?

harsco-jfernandez, 2020-05-18:
The latest Xavier L4T comes with Cuda 10.2. Can tkDNN work with Cuda 10.2 or it must me Cuda 10?

I get the following error and suspect it is because CUDNN v7.605(Cuda 10.2)

./test_yolo3
New NETWORK (tkDNN v0.4, CUDNN v7.605)
CUDNN failure: CUDNN_STATUS_NOT_INITIALIZED
/tkDNN/src/Network.cpp:56
Aborting...

ceccocats, 2020-05-19:
It feels like wrong CUDNN version.
Is it a fresh install from jetpack?
I haven't tested yet with jetpack 4.4 but 4.3 is fine.

harsco-jfernandez, 2020-05-20:
thank you @ceccocats .. yep it was cudnn and cmake as well. it is working now.

I appreciate your effort. viva Italia!! :)

AleBasso80, 2020-05-27:
Hello @ceccocats,
do you think you are going to support jetpack 4.4? Unfortunately this is the only available jetpack for the new Jetson Xavier NX and I think that your project, thanks to the int8 inference, could allow users to get tremendous benefits.
Thanks!

ceccocats, 2020-05-27:
Hello, I can't test right now with the new jetpack but I think it should work.
Looks like @harsco-jfernandez had another problem, can you confirm?

AleBasso80, 2020-05-27:
There is (at least) one compilation error due to cudnn that I don't get with the previous jetpack (that I can't install in the new board). I'm going to investigate the issue, even if I don't master the cuda environment. I'll let you know.

AleBasso80, 2020-05-27:
Hi,
I found the problem. I've already tested with fp32 and fp16 and it works. Tomorrow I'm trying with int8, but it should be fine.
In the file LSTM.cpp at line 89, the function cudnnSetRNNDescriptor must be changed with cudnnSetRNNDescriptor_v6 (cuDNN 8).

ceccocats, 2020-05-27:
Thanks, Nvidia likes changing APIs.

harsco-jfernandez, 2020-05-27:
Nice find @AleBasso80 ...

I've not yet jumped to my Jetson, but surely I'd be running into that. I do have the latest jetpack loaded.  I loaded it like two weeks ago. I'll report back if I run into this as well.

AleBasso80, 2020-05-28:
Yeah, NVIDIA changes to often for my taste...
By the way, I can confirm you that INT8 inference properly works as well.

harsco-jfernandez, 2020-05-29:
yep.. i ran into the api change as well and your change worked. thanks @AleBasso80 

one question on INT8.. are the label files for the calibration the same as the darknet labels?

AleBasso80, 2020-05-29:
@harsco-jfernandez: I've never tried with darknet, but you must use the labels that you find in coco.names or in voc.names, depending the training set you are using.

harsco-jfernandez, 2020-05-30:
I looked at the source to find where the labels are used and it seems they are not. Only the images are loaded into the GPU. It seems that file can have anything in it, as long as it has as many lines as there are images.

AleBasso80, 2020-05-30:
Image file
/.../.../img1.jpg
/.../.../img2.jpg
....
/.../.../imgn.jpg

Label file:
person
cat
...
person

I don't know how to manage images with more detections, therefore I use images where only one targer is detected.

harsco-jfernandez, 2020-05-30:
@AleBasso80 Thank you!

It doesn't matter it seems, the calibration consumes the entire image.

ceccocats, 2020-06-09:
jetpack 4.4 is now supported on master
https://github.com/ceccocats/tkDNN/commit/20303ac32ef8e28ca0ff94be633478d7755ae307

AleBasso80, 2020-06-10:
Hi @ceccocats , did you modify just the function call that I pointed out or did you make other changes/improvements?

Thanks!

ceccocats, 2020-06-10:
Only your changes, but I tested on the new jetpack

closed 2020-06-25 by ceccocats

minhtcai, 2020-07-22:
Hi, I got the same issue while running this: 

./test_yolo3
Not supported field: batch=32
Not supported field: subdivisions=32
Not supported field: momentum=0.9
Not supported field: decay=0.0005
Not supported field: angle=0
Not supported field: saturation = 1.5
Not supported field: exposure = 1.5
Not supported field: hue=.1
Not supported field: learning_rate=0.001
Not supported field: burn_in=1000
Not supported field: max_batches = 500200
Not supported field: policy=steps
Not supported field: steps=400000,450000
Not supported field: scales=.1,.1
New NETWORK (tkDNN v0.5, CUDNN v7.605)
CUDNN failure: CUDNN_STATUS_NOT_INITIALIZED
/home/minh/tkDNN/src/Network.cpp:56
Aborting...

#################
My system:
Ubuntu 18.04
GTX970M
CUDA 10.0
CUDNN 7.6.5
OpenCV 4.4
TensorRT 7.0.0-1+cuda10.0 (install using Debian https://docs.nvidia.com/deeplearning/tensorrt/install-guide/index.html#installing)

Any idea how to fix this? Thank you very much!

mive93, 2020-07-22:
Hi @minhtcai 
try rebooting :)

minhtcai, 2020-07-22:
Hi @mive93 
It works, thank you very much!
I just wondering why did it happen?

mive93, 2020-07-29:
Hi @minhtcai 
Sorry, I don't know, sometimes it happens. 
But the first rule of computer science always helps ;)

zpge, 2020-11-11:
I got the same error
$ ./test_yolo4
New NETWORK (tkDNN v0.5, CUDNN v7.605)
CUDNN failure: CUDNN_STATUS_NOT_INITIALIZED
/data/zpge/tkDNN-master/src/Network.cpp:56
Aborting...

I used the docker from the author---https://hub.docker.com/r/ceccocats/tkdnn
Unfortunately I cannot reboot the computer because it's working as a server. Really don't know how to fix it.

ceccocats, 2020-11-11:
If you are using docker is probably a driver issue. Try `nvidia-smi` and check if your drivers are working. Also try to run a cuda sample just to be sure cuda works

zpge, 2020-11-11:
> If you are using docker is probably a driver issue. Try `nvidia-smi` and check if your drivers are working. Also try to run a cuda sample just to be sure cuda works

Thanks for your reply. nvidia-smi works and a cuda sample from https://gist.github.com/dpiponi/1502434 also works. I also tried another computer and it has the same error as mentioned above.

ceccocats, 2020-11-17:
can you check the versions: https://github.com/ceccocats/tkDNN/issues/143#issuecomment-724269903

lilong-epfl, 2020-12-11:
I tried the latest Jetson Xavier NX L4T with cuda 10.2 and cudnn 8.0.0. I use the docker pulled from NVIDIA website 
[image]

But I still get the same error:
$ ./test_yolo3
New NETWORK (tkDNN v0.5, CUDNN v8)
CUDNN failure: CUDNN_STATUS_NOT_INITIALIZED
/home/xxxx/tkDNN/src/Network.cpp:56
Aborting...

Any idea? Thank you very much!

vodan37, 2021-12-27:
Same error in container