ceccocats/tkDNN #16: Resolution change - Error reading file yolo3_tiny/layers/c15.bin with n of float: 130560 seek: 0 size: 522240

pullmyleg, 2020-05-26:
Hi, any help on the below would be really appreciated. 

I have adjusted the resolution and the number of classes on yolov3_tiny to match trained weights and yolo_v3_tiny config I have (see below) but am getting the following error when running : tane@xavier:~/git/tkDNN/build$ ./test_yolo3_tiny 
Error reading file yolo3_tiny/layers/c15.bin with n of float: 130560 seek: 0 size: 522240

[yolo3_tiny.cpp.txt](https://github.com/ceccocats/tkDNN/files/4680874/yolo3_tiny.cpp.txt)
[yolo3_tiny.cfg.txt](https://github.com/ceccocats/tkDNN/files/4681052/yolo3_tiny.cfg.txt)

Error screen: 
[image]

1. I can successfully run example tests e.g. yolov4 and demo with no issue. 
2. I can confirm the configuration used in training of weights is identical to yolo3_tiny.cfg sample provided in this repo apart from class, resolution, and filters change. 
3. Adjustments are made before compiling (make clean, make) the code. 
4. Weights files (successfully?) converted using the guide provided in the readme. output here. 
[yolo3_tiny_weights_convert.txt](https://github.com/ceccocats/tkDNN/files/4681091/yolo3_tiny_weights_convert.txt)

I am attempting to do a performance comparison of trained models darknet vs tkDNN which will run on a Xavier AGX onboard a drone to track critically endangered dolphins. The better the FPS the better overall tracking performance.

Thanks!

mive93, 2020-05-26:
hi @pullmyleg,

I see at least 3 issues:
  * ```tk::dnn::dataDim_t dim(1, 3, 1920, 1088, 1)``` is wrong and should be ```tk::dnn::dataDim_t dim(1, 3, 1088, 1920, 1)``` because it follows the NCHW pattern.
  * in your config file you have classes = 1 (line 135) and classes=80 (line 177). Is it what you wanted? 
  * in your configuration file the number of filters before yolo layers are 18, however in the cpp there is 255 (lines 67,82). E.g.``` tk::dnn::Conv2d     c15(&net, 255, 1, 1, 1, 1, 0, 0, c15_bin, false);``` should be ``` tk::dnn::Conv2d     c15(&net, 18, 1, 1, 1, 1, 0, 0, c15_bin, false);```

I think that by solving those errors you should be able to run it correctly. Let me know if you still have problems.

pullmyleg, 2020-05-26:
Thanks @mive93 , especially for just a quick response! 

Changes made as per suggestions. Got a little bit further :). 

It is now getting stuck at c22.bin - Error reading file yolo3_tiny/layers/c22.bin with n of float: 20992 seek: 0 size: 83968

[image]

[yolo3_tiny.cfg.txt](https://github.com/ceccocats/tkDNN/files/4681389/yolo3_tiny.cfg.txt)
[yolo3_tiny.cpp.txt](https://github.com/ceccocats/tkDNN/files/4681390/yolo3_tiny.cpp.txt)

Thanks again for your help!

mive93, 2020-05-26:
Hi @pullmyleg 

I think it is still a problem with the number of filters before the last yolo layer.
```    tk::dnn::Conv2d     c22(&net, 82, 1, 1, 1, 1, 0, 0, c22_bin, false);```  (line 82) should be     ```tk::dnn::Conv2d     c22(&net, 18, 1, 1, 1, 1, 0, 0, c22_bin, false);```

pullmyleg, 2020-05-26:
Sorry & thanks for your help @mive93, how stupid! It's working now. :facepalm:

I will post the performance result comparison tomorrow after some sleep!

pullmyleg, 2020-06-01:
Results are great thank you! 
~**32FPS** on **tkDNN** FP16 (this repo) vs **22FPS** on [**Darknet** (AlexyAB)](url). 

Final configs below if anyone is interested.

[yolo3_tiny.cpp.txt](https://github.com/ceccocats/tkDNN/files/4709174/yolo3_tiny.cpp.txt)
[yolo3_tiny.cfg.txt](https://github.com/ceccocats/tkDNN/files/4709176/yolo3_tiny.cfg.txt).

Now I am attempting to build this 3 layer yolo tiny model: [yolo_v3_tiny_pan3_scale_giou-sml.cfg.txt](https://github.com/ceccocats/tkDNN/files/4709200/yolo_v3_tiny_pan3_scale_giou-sml.cfg.txt) in tkDNN because I get much greater accuracy [+10% mAP.](https://github.com/AlexeyAB/darknet/issues/5552) vs yolo3_tiny but to date with a much lower frame rate.

closed 2020-06-01 by pullmyleg