ceccocats/tkDNN #19: How sholud I add a new layer?

beizhengren, 2020-05-28:
1.How sholud I add a new layer with tkDNN?
2.How to manual generating trt model?
I use the command
```
./darknet export <path-to-cfg-file> <path-to-weights> layers
./darknet export <path-to-cfg-file> <path-to-weights> debug
```
And 
```
./test_yolo3            # run the yolo test (is slow)
```
auto generate yolo3_fp32.rt with exported bin files in cloud.

ceccocats, 2020-05-28:
Hi @beizhengren,
1. What do you mean? Adding a new kind of layer or adding a new layer in a model?

2. For example if you use test_yolo3 then the weights will be downloaded from the cloud and put into yolo3 in your build folder. If you want to use your exported weights, override the downloaded weights inside yolo3 folder.

beizhengren, 2020-05-29:
Hi @ceccocats,
Thanks for your quick reply.
1. A new kind of layer. Concretely, I have a project that can load tensorrt model or darknet files(cfg, weights) to inference. And there are some new layer tensorrt can not parse, so I want to use tkDNN
to help solve it, for example yolov4.  That is embedding tkDNN to my own project. 

2. Some “Wrongs” prompted. My work flow as following
I use the command
```sh
 ./darknet export cfg/yolov4.cfg yolov4.weights layers debug 
```
and files of debug folder, as following
[image]

layers  folder as following
[image]
then run
```
./test_yolo4   
```
<details>
<summary>please click</summary>

```
$ ./test_yolo4                                                  [18411d 0h 12m]
New NETWORK (tkDNN v0.4, CUDNN v7.5)
Reading weights: I=3 O=32 KERNEL=3x3x1
Reading weights: I=32 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=32 KERNEL=1x1x1
Reading weights: I=32 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=64 O=128 KERNEL=3x3x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=3x3x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=1x1x1
Reading weights: I=256 O=512 KERNEL=3x3x1
Reading weights: I=512 O=256 KERNEL=1x1x1
Reading weights: I=512 O=256 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=3x3x1
Reading weights: I=256 O=256 KERNEL=1x1x1
Reading weights: I=256 O=256 KERNEL=3x3x1
[... 417 lines cut ...]
CUDNN vs correct
 | [ 0 ]: 0.883185 0.662702
 | [ 1 ]: 0.537623 0.44748
 | [ 2 ]: 0.419132 0.250894
 | [ 3 ]: 0.512994 0.843529
 | [ 4 ]: 0.394617 0.248704
 | [ 5 ]: 0.543686 0.20504
 | [ 6 ]: 0.56079 0.85909
 | [ 7 ]: 0.362244 0.754667
 | [ 8 ]: 0.568176 0.775216
 | Wrongs: 9748 ~0.02
TRT   vs correct
 | [ 0 ]: 0.883186 0.662702
 | [ 1 ]: 0.537623 0.44748
 | [ 2 ]: 0.419132 0.250894
 | [ 3 ]: 0.512994 0.843529
 | [ 4 ]: 0.394617 0.248704
 | [ 5 ]: 0.543685 0.20504
 | [ 6 ]: 0.560791 0.85909
 | [ 7 ]: 0.362245 0.754667
 | [ 8 ]: 0.568176 0.775216
 | Wrongs: 9748 ~0.02

```
</details>

ceccocats, 2020-05-29:
To add a new layer you need to:
- insert it in Layers.h and implement all the functions of it, you can use other similar layer as inspiration. Also add your layer to ther layers list in layerType_t and the printing of it in the Layer::getLayerName().
If you do it well you will get the inference for CUDNN.
- add a tensorrt layer in NetworkRT.cpp specialising the NetworkRT::convert_layer function. If your layer is supported by tensorrt it will be easy, if not you need to do a plug-in, some examples in pluginsRT folder.

beizhengren, 2020-05-29:
@ceccocats 
1. Thanks for your detailed answer I will try it.
2. The cfg and weights are downloaded from  https://github.com/AlexeyAB/darknet
>If the cfg is different from the one in the test folder you need to change the code.  

That means I should put my cfg, which corresponding to my own weights, in the`tkDNN-master/tests/yolo4` folder and replace the existed `yolov4.cfg`?

Here is my yolov4 cfg
<details>
<summary>please click </summary>

```sh
[net]
batch=64
subdivisions=8
# Training
#width=512
#height=512
width=608
height=608
channels=3
momentum=0.949
decay=0.0005
angle=0
saturation = 1.5
exposure = 1.5
hue=.1

learning_rate=0.00261
burn_in=1000
max_batches = 500500
policy=steps
steps=400000,450000
scales=.1,.1

#cutmix=1
mosaic=1

#:104x104 54:52x52 85:26x26 104:13x13 for 416

[convolutional]
batch_normalize=1
filters=32
size=3
stride=1
pad=1
activation=mish

# Downsample

[convolutional]
batch_normalize=1
filters=64
size=3
stride=2
pad=1
activation=mish

[convolutional]
batch_normalize=1
filters=64
size=1
stride=1
pad=1
activation=mish

[route]
layers = -2

[convolutional]
[... 1061 lines cut ...]
[convolutional]
size=1
stride=1
pad=1
filters=255
activation=linear

[yolo]
mask = 6,7,8
anchors = 12, 16, 19, 36, 40, 28, 36, 75, 76, 55, 72, 146, 142, 110, 192, 243, 459, 401
classes=80
num=9
jitter=.3
ignore_thresh = .7
truth_thresh = 1
random=1
scale_x_y = 1.05
iou_thresh=0.213
cls_normalizer=1.0
iou_normalizer=0.07
iou_loss=ciou
nms_kind=greedynms
beta_nms=0.6
```
</details>

ceccocats, 2020-05-29:
Hi, our test input size is 416x416, yours apper to be 608x608.
You need to change the input dim of the network in yolo4.cpp. (line 9)
Also check issue #10 you probably will get NaNs but is not an issue.

beizhengren, 2020-06-12:
Hi, @ceccocats 
Sorry for I did't reply until now. I am so busy recently.
1. when I using the command:
```sh
./darknet export cfg/yolov4.cfg yolov4.weights layers debug
```
There ares a lot of files in layers
```
c0.bin    c117.bin  c132.bin  c146.bin  c159.bin  c27.bin  c41.bin  c56.bin  c6.bin   c86.bin  g139.bin
c100.bin  c11.bin   c133.bin  c147.bin  c15.bin   c28.bin  c43.bin  c58.bin  c71.bin  c87.bin  g150.bin
c102.bin  c120.bin  c134.bin  c148.bin  c160.bin  c29.bin  c44.bin  c59.bin  c72.bin  c89.bin  g161.bin
c104.bin  c122.bin  c135.bin  c149.bin  c16.bin   c2.bin   c46.bin  c5.bin   c74.bin  c8.bin   input.bin
c105.bin  c123.bin  c136.bin  c14.bin   c18.bin   c31.bin  c47.bin  c60.bin  c75.bin  c90.bin  output.bin
c106.bin  c124.bin  c137.bin  c152.bin  c19.bin   c32.bin  c49.bin  c62.bin  c77.bin  c91.bin
c107.bin  c125.bin  c138.bin  c154.bin  c1.bin    c34.bin  c4.bin   c63.bin  c78.bin  c93.bin
c10.bin   c126.bin  c141.bin  c155.bin  c21.bin   c35.bin  c50.bin  c65.bin  c80.bin  c94.bin
c114.bin  c127.bin  c143.bin  c156.bin  c23.bin   c37.bin  c52.bin  c66.bin  c81.bin  c96.bin
c115.bin  c12.bin   c144.bin  c157.bin  c24.bin   c38.bin  c54.bin  c68.bin  c83.bin  c97.bin
c116.bin  c130.bin  c145.bin  c158.bin  c25.bin   c40.bin  c55.bin  c69.bin  c85.bin  c99.bin
```
and  none is in debug.

2. And how can I make this binary files to .trt file ?
my cfg and weights are from  https://github.com/AlexeyAB/darknet

Thanks!

ceccocats, 2020-06-12:
1. To get debug you must compile the darknet fork without GPU
2. Create a new test starting from our yolo4

beizhengren, 2020-06-15:
@ceccocats 
I’ll try it.
Thank you!

closed 2020-09-11 by mive93