ceccocats/tkDNN #20: CUDNN_NVLIB issue

ixtiyoruz, 2020-05-29:
Hi,
When I try to make the project it is throwing an error like:
```

-- Found NVINFER: CUDNN_NVLIB-NOTFOUND
install dir:/usr/local
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDNN_NVLIB
    linked by target "tkDNN" in directory /home/ixtiyor/Documents/projects/temp/tkDNN

```

I have tried different versions of tensorrt but it is not working, can somebody help me please ?

ceccocats, 2020-05-29:
You need to find where you installed libnvinfer.so and the fix the file:
https://github.com/ceccocats/tkDNN/blob/master/cmake/FindCUDNN.cmake

ixtiyoruz, 2020-05-29:
> You need to find where you installed libnvinfer.so and then fix the file:
> https://github.com/ceccocats/tkDNN/blob/master/cmake/FindCUDNN.cmake

That worked, thank you very much.

closed 2020-05-29 by ixtiyoruz

ixtiyoruz, 2020-05-29:
Sorry, but I have another problem, during make : 
```
tkDNN/include/tkDNN/NetworkRT.h:8:21: fatal error: NvInfer.h: No such file or directory
compilation terminated.
CMake Error at kernels_generated_postprocessing.cu.o.cmake:207 (message):
  Error generating
  /home/ixtiyor/Documents/projects/temp/tkDNN/build/CMakeFiles/kernels.dir/src/kernels/./kernels_generated_postprocessing.cu.o

CMakeFiles/kernels.dir/build.make:77: recipe for target 'CMakeFiles/kernels.dir/src/kernels/kernels_generated_postprocessing.cu.o' failed
make[2]: *** [CMakeFiles/kernels.dir/src/kernels/kernels_generated_postprocessing.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:713: recipe for target 'CMakeFiles/kernels.dir/all' failed
make[1]: *** [CMakeFiles/kernels.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```
I defined the findpath like : 

```

find_path(CUDNN_INCLUDE_DIR
        ${CMAKE_SYSROOT}/usr/local/include
        ${CMAKE_SYSROOT}/usr/include
        /home/ixtiyor/Downloads/cudnn_all/new_cudnn_all/TensorRT-6.0.1.5/include
        /usr/local/nvidia/tensorrt/include/
        NO_DEFAULT_PATH
)

```

Can you please tell me what is missing here ?

ixtiyoruz, 2020-05-29:
> Sorry, but I have another problem, during make :
>
> ```
> tkDNN/include/tkDNN/NetworkRT.h:8:21: fatal error: NvInfer.h: No such file or directory
> compilation terminated.
> CMake Error at [...]

I solved it by adding  
`include_directories("/home/ixtiyor/Downloads/cudnn_all/new_cudnn_all/TensorRT-6.0.1.5/include/")`