ceccocats/tkDNN #23: Where are the models?

zpmmehrdad, 2020-05-30:
Hi,

I could compile it on windows and I couldn't find the models?
Where are the models? for example yolo3_berkeley.rt

Thanks

zpmmehrdad, 2020-05-30:
I found it

closed 2020-05-30 by zpmmehrdad

xjsxujingsong, 2020-06-03:
@zpmmehrdad  I am working on windows as well.
I can complie to get kernels.lib. but when compile tkDNN, it always show cannot find the functions:

1>------ Build started: Project: tkDNN, Configuration: Release x64 ------
1>   Creating library D:/code/yolo/tkDNN/build2/Release/tkDNN.lib and object D:/code/yolo/tkDNN/build2/Release/tkDNN.exp
1>Activation.obj : error LNK2019: unresolved external symbol "void __cdecl activationLEAKYForward(float *,float *,int,struct CUstream_st *)" (?activationLEAKYForward@@YAXPEAM0HPEAUCUstream_st@@@Z) referenced in function "public: virtual float * __cdecl tk::dnn::Activation::infer(struct tk::dnn::dataDim_t &,float *)" (?infer@Activation@dnn@tk@@UEAAPEAMAEAUdataDim_t@23@PEAM@Z)
1>NetworkRT.obj : error LNK2001: unresolved external symbol "void __cdecl activationLEAKYForward(float *,float *,int,struct CUstream_st *)" (?activationLEAKYForward@@YAXPEAM0HPEAUCUstream_st@@@Z)
1>Activation.obj : error LNK2019: unresolved external symbol "void __cdecl activationMishForward(float *,float *,int,struct CUstream_st *)" (?activationMishForward@@YAXPEAM0HPEAUCUstream_st@@@Z) referenced in function "public: virtual float * __cdecl tk::dnn::Activation::infer(struct tk::dnn::dataDim_t &,float *)" (?infer@Activation@dnn@tk@@UEAAPEAMAEAUdataDim_t@23@PEAM@Z)

I have double checked in tkDNN project, the library Release/kernels.lib is there....
Any suggestions?

zoom1539, 2020-08-27:
@xjsxujingsong I met the same error, have you solved it?