ceccocats/tkDNN #25: How to change the classes of models

weichiheng, 2020-06-01:
Hello, thank you for your work!
I trained yolo3 with a dataset which only have 4 classes. After I changing the files in debug and layers, I try to use test_yolo3 to create the rt file. Then I got an error:
```
Error reading file yolo3/layers/c81.bin with n of float: 261120 seek: 0 size: 1044480
```
I know the reason is the different between classes.
In /tkDNN/include/tkDNN/models/Yolo3.h , I can find:
```
int preYoloFilters = (classes+5)*3; 
```
How can I change the classes?
Thank you!

ceccocats, 2020-06-01:
Now you can directly parse the cfg from darknet
https://github.com/ceccocats/tkDNN#darknet-parser

weichiheng, 2020-06-02:
@ceccocats Thank you for your reply!
I have looked over the scheme you gave, but the problem has not been solved yet.
```tk::dnn::Network *net```is not a ```tk::dnn::NetworkRT``` instance, can I still use TensorRT to speed up inference?
I just want to convert the yolo network which has changed the number of classes into an rt model,  then run the demo with the rt model, can you show me how to make it?
Thank you very much!

ceccocats, 2020-06-02:
Test file use both network and networkRT

weichiheng, 2020-06-07:
> Test file use both network and networkRT

Thank you for your reply! I was doing something else the other day, but today I gave it a try and success.

closed 2020-06-07 by weichiheng