ceccocats/tkDNN #28: engineRT is NULL after deserialize

zpmmehrdad, 2020-06-03:
Hi,

I compiled it on Windows and when I have serialized the model successfully but when it wants to deserialize the engineRT is null. Where is the problem?

Thanks

ceccocats, 2020-06-04:
What mode are you using? FP 32, 16 or int8?
Please activate the debug mode to see if there are some problems:
```
cmake .. -DDEBUG=True
make
```

```

zpmmehrdad, 2020-06-05:
@ceccocats Hi,

I'm using FP 32 and Tensor rt v7, the serialization is worked perfectly but when I want to run with .rt I get the error

ceccocats, 2020-06-05:
I need more info, does it print something with debug active? Can you paste the output? Does it create the .rt file?

oessam, 2020-06-07:
@zpmmehrdad  Hello,
Could you please post any porting hints to compile it on windows ? 

Thank you

oessam, 2020-06-07:
@zpmmehrdad  What shall be the configuration type ( Application .exe, static lib, dll ) ?

zpmmehrdad, 2020-06-07:
> @zpmmehrdad What shall be the configuration type ( Application .exe, static lib, dll ) ?

@oessam Hi,

I have added the files to my project and compile it. I could test it successfully but when after that use the *.rt files I get an error.

oessam, 2020-06-07:
@zpmmehrdad  didnt you use cmake to generate a project ? or how was your project ready to be compiled ?

zpmmehrdad, 2020-06-07:
> @zpmmehrdad didnt you use cmake to generate a project? 

Yes exactly, I didn't use CMake

ceccocats, 2020-06-07:
If you can, use cmake... Cmake is love, cmake is life

zpmmehrdad, 2020-06-08:
@ceccocats Hi,

I don't think my problem is CMake

JoeCool90, 2020-06-14:
I'm guessing it was the binary file problem.  Make sure both the serialising and deserialising is done on a file that has been opened in binary mode.

dsgh2, 2020-09-22:
How did you build on Windows? Please let me know how to build