ceccocats/tkDNN #29: yolov3-spp？

ou525, 2020-06-04:
Thank you very much for your work.
Have you tested the effect of yolov3-spp network deployment on tkDnn, is it consistent with darknet?

ceccocats, 2020-06-04:
If yolov3-spp use the same layers and parameters of yolo3 it is supported by our Darknet cfg parser

ou525, 2020-06-05:
I have converted my yolov3-spp network successfully (only modify the category and my own dataset), and tested some pictures, and found that some output is not consistent.
I don't know where is the problem

ceccocats, 2020-06-05:
Can you list all the steps you have done?
You should:
- get your net.cfg and net.weights in darknet
- export the weights using our darknet fork
- make a test in tkdnn or modify an existing one that take your net.cfg and your exported weights.
- execute the test and check if it is correct
- use the generated rt file in demo or whatsoever

ou525, 2020-06-05:
Yes, I have completed all the steps, otherwise I cannot continue.
new a test cpp
#include<iostream>
#include<vector>
#include "tkdnn.h"
#include "test.h"
#include "DarknetParser.h"

int main() {
    std::string bin_path  = "yolo_spp";
    std::vector<std::string> input_bins = { 
        bin_path + "/layers/input.bin"
    };
    std::vector<std::string> output_bins = {
        bin_path + "/debug/layer89_out.bin",
        bin_path + "/debug/layer101_out.bin",
        bin_path + "/debug/layer113_out.bin"
    };
    std::string wgs_path  = bin_path + "/layers";
    std::string cfg_path  = "my.cfg";
    std::string name_path = "my.txt";
    downloadWeightsifDoNotExist(input_bins[0], bin_path, "https://cloud.hipert.unimore.it/s/d97CFzYqCPCp5Hg/download");

    // parse darknet network
    tk::dnn::Network *net = tk::dnn::darknetParser(cfg_path, wgs_path, name_path);
    net->print();

    //convert network to tensorRT
    tk::dnn::NetworkRT *netRT = new tk::dnn::NetworkRT(net, net->getNetworkRTName(bin_path.c_str()));
    
    int ret = testInference(input_bins, output_bins, net, netRT);
    net->releaseLayers();
    delete net;
    delete netRT;
    return ret;
}

ceccocats, 2020-06-05:
Are you sure the output layers are the same?  If you add or remove layers the output number will change.
To be sure you can set this as output-bins: 
```
std::vectorstd::string output_bins = {
bin_path + "/debug/layer89_out.bin",
bin_path + "/debug/layer101_out.bin",
bin_path + "/layers/output.bin"
};
```
Output.bin is the last layer output, so at least the last yolo layer should be right.
For the others yolos you must find the correct number when you are exporting from darknet

ou525, 2020-06-06:
The output layer should be correct, I did not modify the network structure,

ceccocats, 2020-06-06:
The nms affect the results only if you do the postprocessing that is not executed in the test.
Could you paste the cfg? Maybe also weights.
When I have this problems i remove layers until I find what is the layer that create the error.
Maybe your cfg contains an option not parsed.

ou525, 2020-06-08:
thank有，i uploaded the cfg and weights files, you can use it as a reference test.
https://drive.google.com/file/d/10vGFfsRjCrP-jKU7L5vrkjZtBF-gV5LN/view?usp=sharing
https://drive.google.com/file/d/1eUq03hKS_po7HDk6X67QSFr10t3xYizf/view?usp=sharing

NickiBD, 2020-06-08:
@ceccocats ,Hi,
Thanks lot for your great work .I had a question regarding execution of the changed .cpp for parser .I have created a new cpp file (parser ) and changed the out puts for the custom cfg in darknet (yolov3-spp) and also exported the weights (bin files ). However, how should I execute the changed .cpp file to generate the rt file for the demo.  I used  g++  ..... to compile the .cpp file but  it gave me include errors (cuda.h , tkdnn no such file or directory ) .I would be really grateful if you could help me with this . Thanks in advance .

ceccocats, 2020-06-12:
> thank you，i uploaded the cfg and weights files, you can use it as a reference test.
> https://drive.google.com/file/d/10vGFfsRjCrP-jKU7L5vrkjZtBF-gV5LN/view?usp=sharing
> https://drive.google.com/file/d/1eUq03hKS_po7HDk6X67QSFr10t3xYizf/view?usp=sharing

I still can't access the files

ou525, 2020-06-13:
sorry，please try again
https://drive.google.com/file/d/1eUq03hKS_po7HDk6X67QSFr10t3xYizf/view?usp=sharing
https://drive.google.com/file/d/10vGFfsRjCrP-jKU7L5vrkjZtBF-gV5LN/view?usp=sharing

ceccocats, 2020-06-14:
I executed it without any problems, but pull the repo, some tensorRT issue can be due to different include and library dirs, the last commits solves this cmake issue.
I changed the output bins that in your network are different, the simplier way to check what de bug layers are the YOLO ones is to check what layers weights starts with "g":
```
build$ ls yolov3-sppx/layers/g*
yolov3-sppx/layers/g101.bin  yolov3-sppx/layers/g113.bin  yolov3-sppx/layers/g89.bin
```

This is the code:
```
#include<iostream>
#include<vector>
#include "tkdnn.h"
#include "test.h"
#include "DarknetParser.h"

int main() {
    std::string bin_path  = "yolov3-sppx";
    std::vector<std::string> input_bins = { 
        bin_path + "/layers/input.bin"
    };
    std::vector<std::string> output_bins = {
        bin_path + "/debug/layer89_out.bin",
        bin_path + "/debug/layer101_out.bin",
        bin_path + "/debug/layer113_out.bin"
    };
    std::string wgs_path  = bin_path + "/layers";
    std::string cfg_path  = "../tests/darknet/cfg/yolov3-sppx.cfg";
    std::string name_path = "../tests/darknet/names/coco4.names";
    //downloadWeightsifDoNotExist(input_bins[0], bin_path, "https://cloud.hipert.unimore.it/s/jPXmHyptpLoNdNR/download");

    // parse darknet network
    tk::dnn::Network *net = tk::dnn::darknetParser(cfg_path, wgs_path, name_path);
    net->print();

    //convert network to tensorRT
    tk::dnn::NetworkRT *netRT = new tk::dnn::NetworkRT(net, net->getNetworkRTName(bin_path.c_str()));
    
    int ret = testInference(input_bins, output_bins, net, netRT);    
    net->releaseLayers();
    delete net;
    delete netRT;
    return ret;
}
```

ou525, 2020-06-15:
@ceccocats sorry, this code seems to be basically the same as mine

closed 2020-09-11 by mive93