ceccocats/tkDNN #35: Darknet export bugs

JoeCool90, 2020-06-14:
Hi, me again.

In run_export in darknet.c in your other rep (sorry, don't know how to post issues on that one) there are a couple of bugs that caused me problems.

* The fopen's need to be in binary mode
* char *f[256] should be char f[256] or maybe char f[MAX_PATH] or whatever crossplatform version would work

Thanks for all your work btw.

ceccocats, 2020-06-14:
> * The fopen's need to be in binary mode
Same error... Does it create problems in Windows? Because in GNU/Linux never had any issue with this.
> * char *f[256] should be char f[256] or maybe char f[MAX_PATH] or whatever crossplatform version would work
Actually I found this problem this morning and substitute with 2048 but yeah, MAX_PATH is better.

Thank you

JoeCool90, 2020-06-22:
Sorry, didn't spot your reply.

Yes the error occurred with Visual Studio on Windows.