ceccocats/tkDNN #38: Test inference time with batch_size != 1

DoriHp, 2020-06-15:
Thank for your greate work. Currently, I'm trying to find out the different between inference time of running with batch size 1 and 4 for fp16 model. Here my test script (name ```image.cpp```):
```
#include <iostream>
#include <signal.h>
#include <stdlib.h>     
#include <unistd.h>
#include <mutex>

#include "Yolo3Detection.h"
#include <chrono>

using namespace std::chrono; 

int main(int argc, char *argv[]){

    // Network name
    std::string net = "yolo3-tiny_3l.rt";
    if(argc > 1)
        net = argv[1];
    
    // Inference batch size 
    int n_batch = 4;
    if(argc > 2)
        n_batch = atoi(argv[2]);
        
    // Number of classes
    int n_classes = 2;
    if(argc > 3)
        n_classes = atoi(argv[3]);
  
    tk::dnn::Yolo3Detection yolo;  

    tk::dnn::DetectionNN *detNN; 
    detNN = &yolo;
    detNN->init(net, n_classes, n_batch);	
    
    cv::Mat img;
    std::vector<cv::Mat> batch_frame;
    std::vector<cv::Mat> batch_dnn_input;
    auto start = high_resolution_clock::now(); 
    // cv::namedWindow("detection", cv::WINDOW_NORMAL);  

    // Read all images from a directory          
    cv::String path("../images/*.*");
    std::vector<cv::String> fn;
    cv::glob(path,fn,true); // recurse
    int total = 0;
    int num_img = 0; 
    for (size_t k=0; k < fn.size(); ++k){
         img = cv::imread(fn[k], cv::IMREAD_COLOR);
	 if (!img.data){
             std::cout << "Problem loading image!";
             break;
         } 
         batch_frame.push_back(img);

         // this will be resized to the net format
         batch_dnn_input.push_back(img.clone());
         ++num_img;
          
         
         // do inference
         if (num_img == n_batch){
             detNN->update(batch_dnn_input, n_batch);
             detNN->draw(batch_frame); 
             for(int bi=0; bi< n_batch; ++bi){
             //    cv::imshow("detection", batch_frame[bi]);
             //    cv::waitKey(10);
                   ++total;
             }           
             num_img = 0;
             batch_dnn_input.clear();
             batch_frame.clear();
         }            
    }
    auto stop = high_resolution_clock::now();
    auto duration = duration_cast<seconds>(stop - start);  
    double mean = 0; 
    std::cout<<COL_GREENB<<"\n\nTime stats:\n";
    std::cout << "Total files: " << total << " files\n";
    std::cout << "Total inference time: " << duration.count() << "s\n";
    std::cout<<"Min: "<<*std::min_element(detNN->stats.begin(), detNN->stats.end())/n_batch<<" ms\n";    
    std::cout<<"Max: "<<*std::max_element(detNN->stats.begin(), detNN->stats.end())/n_batch<<" ms\n";    
    for(int i=0; i<detNN->stats.size(); i++) mean += detNN->stats[i]; mean /= detNN->stats.size();
    std::cout<<"Avg: "<<mean/n_batch<<" ms\t"<<1000/(mean/n_batch)<<" FPS\n"<<COL_END; 
    
}
```
What I did to generate my model:
```
export TKDNN_MODE=FP16
export TKDNN_BATCHSIZE=4
./test_yolo3tiny_512
```
and this one to compare:
```
export TKDNN_MODE=FP16
export TKDNN_BATCHSIZE=1
./test_yolo3tiny_512
```
But when running test case, this is what I received:
For batch_size = 1:
use this command:
```
./image yolo3-tiny_512_fp16.rt 1
```
and this is result:
```
New NetworkRT (TensorRT v5.16)
Float16 support: 1
Int8 support: 0
DLAs: 0
create execution context
Input/outputs numbers: 4
input idex = 0 -> output index = 3
Data dim: 1 3 512 512 1
Data dim: 1 21 64 64 1
RtBuffer 0   dim: Data dim: 1 3 512 512 1
RtBuffer 1   dim: Data dim: 1 21 16 16 1
RtBuffer 2   dim: Data dim: 1 21 32 32 1
RtBuffer 3   dim: Data dim: 1 21 64 64 1

Time stats:
Total files: 1000 files
Total inference time: 85s
Min: 42.6142 ms
Max: 2113.66 ms
Avg: 46.6069 ms	21.456 FPS
```

For batch_size = 4:
use this command:
```
./image yolo3-tiny_512_fp16.rt 4
```
and this is result:
```
New NetworkRT (TensorRT v5.16)
Float16 support: 1
Int8 support: 0
DLAs: 0
create execution context
Input/outputs numbers: 4
input idex = 0 -> output index = 3
Data dim: 1 3 512 512 1
Data dim: 1 21 64 64 1
RtBuffer 0   dim: Data dim: 1 3 512 512 1
RtBuffer 1   dim: Data dim: 1 21 16 16 1
RtBuffer 2   dim: Data dim: 1 21 32 32 1
RtBuffer 3   dim: Data dim: 1 21 64 64 1

Time stats:
Total files: 1000 files
Total inference time: 86s
Min: 42.2471 ms
Max: 652.825 ms
Avg: 46.5498 ms	21.4823 FPS
```
As you can see, there is no huge diffenrent as described in your README. What did I do wrong?

mive93, 2020-06-15:
Hi @DoriHp,

I think it's a problem of graphic card. Which model do you have? 
Batching is not magic, it helps if your GPU still have resources to be exploited, but if it's small then it will have similar results as single batch (look at the FPS of the Jetson Nano in the README).

I have run your code on a RTX2080Ti, on 5K mages (COCO2017 val) and this is what I get: 
```
Avg: 1.3413 ms          745.547 FPS
Avg: 0.685971 ms        1457.79 FPS
```

DoriHp, 2020-06-15:
Thank for your reply. Look like it was a problem with gpu memory because I have only tested this script on my Jetson Nano. I will try to run it on Xavier and tell you the result :3

closed 2020-06-15 by DoriHp

DoriHp, 2020-06-16:
Here my test resutl on other platform:
GTX2060 8G:
- FP16, batch_size=1: Avg: 1.81135 ms 552.074 FPS
- FP16, batch_size=4: Avg: 1.11043 ms 900.554 FPS

Jetson Xavier:
- FP32, batch_size=1: avg ~85fps
- FP32, batch_size=4: avg ~124fps

Now I know how GPU memory affects in case batch size changing. Thank you a lot. And one more question, how can I get confidence of bbox? I couldn't find in anywhere @mive93.

mive93, 2020-06-16:
Hi @DoriHp 

Cool that you succeeded :) 
Each bounding box has a prob filed that represent the probability. You can access all the boxes from your main in image.cpp via detNN->batchDetected (for each batch there is a vector of bounding boxes). 
Otherwise just access it in the draw function (in the for in lines 160-175 in DetectionNN.h)

basailuonaaa, 2020-08-20:
Hi @mive93 
I have run the code on a nvidia RTX2080Ti(ubuntu16.04, TensorRT-6.0.1.5, cuda-10.0, cudnn-7.6.1), on 1514 mages and this is what I get:
For YoloV4 608*608 model:

for FP32.rt:
Batch = 1	Avg: 15.0531 ms          66.4315 FPS
In addition, the time of the entire for loop is 42702ms,the average time to process each picture is 28ms;
Batch = 4	Avg: 13.3748 ms        74.7676FPS
In addition, the time of the entire for loop is 40573 ms,the average time to process each picture is 26ms;

for FP16.rt:
Batch = 1	Avg: 8.89959 ms          112.365 FPS
In addition, the time of the entire for loop is 33509 ms, the average time to process each picture is 22 ms;
Batch = 4	Avg: 6.85899 ms        145.794 FPS
In addition, the time of the entire for loop is 31408 ms,the average time to process each picture is 20ms;

for int8.rt:
Batch = 1	Avg: 7.38597 ms          135.392 FPS
In addition, the time of the entire for loop is 30809 ms,the average time to process each picture is 20ms;
Batch = 4	Avg: 5.97115 ms        167.472 FPS
In addition, the time of the entire for loop is 31217 ms,the average time to process each picture is 20ms;

The fps is gradually increasing, but the average time to process each picture is basically the same. Is this a normal phenomenon?