ceccocats/tkDNN #39: CUDA_cublas_device_LIBRARY (ADVANCED)

deep-practice, 2020-06-15:
-- Found CUDNN: /usr/local/cuda/lib64/libcudnn.so
-- Found CUDNN include: /usr/local/cuda/include
-- Found NVINFER: /usr/local/TensorRT-7.0.0.11/lib
-- Found NVINFER include: /usr/local/TensorRT-7.0.0.11/include
install dir:/usr/local
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_cublas_device_LIBRARY (ADVANCED)
    linked by target "tkDNN" in directory /home/work/deep_learning/tkDNN

ceccocats, 2020-06-15:
This is a known issue on Ubuntu 18 apt cmake.
You need to install at least cmake 3.15

deep-practice, 2020-06-16:
Thx.It works.

closed 2020-06-16 by deep-practice

kochsebastian, 2020-09-09:
I'm trying to run it on a Jetson Xavier. For that only Cmake 3.10.2 is availible. Any solutions for that?

mive93, 2020-09-09:
Hi @kochsebastian 
Just download a later version (https://cmake.org/download/), unzip and compile cmake. 
You can compile cmake with cmake itself by doing, in the extracted folder:

```
cmake .
make      
make install
```

kochsebastian, 2020-09-10:
The problem is the Jetson Xavier has an arm architecture and I only see x86_64 versions on the Cmake download page

mive93, 2020-09-15:
That's why you have to compile it. 
Download this: https://github.com/Kitware/CMake/releases/download/v3.18.2/cmake-3.18.2.tar.gz
Unzip it, then move inside the unzipped folder and finally
```
mkdir build
cd build
cmake ..
make -j4
make install
```

rickymedrano, 2021-02-07:
> That's why you have to compile it.
> Download this: https://github.com/Kitware/CMake/releases/download/v3.18.2/cmake-3.18.2.tar.gz
> Unzip it, then move inside the unzipped folder and finally
> 
> ```
> mkdir build
> cd build
> cmake ..
> make -j4
> make install
> ```

For anybody having issues with cmake still on Jetson,I had to run `sudo make install` for last step. Also make sure to uninstall the old cmake first with:
`sudo apt remove --purge --auto-remove cmake`