ceccocats/tkDNN #42: Could not build Cuda Engine

rod-hendricks, 2020-06-16:
I was wondering if anyone could help me out as to what I am missing. When i run ./test_yolo4, my buildEngineWithConfig() does not create a cuda engine for me. There doesn't seem to be any other errors not sure what i am missing. 

GPU: RTX2070 super
CUDA: 10.0
OPENCV: 3.4.7
Cudnn: 7.6.3
TensorRT: 6.0.1

[image]

ceccocats, 2020-06-16:
This error happen often when there is not enough memory, but it is not your case.
Please activate the debug with:
```
cmake .. -DDEBUG=True
make
```

This is what you see when your memory is finished:
```
TENSORRT LOG: ../rtSafe/safeRuntime.cpp (25) - Cuda Error in allocate: 2 (out of memory)
TENSORRT LOG: GPU memory allocation error during getBestTactic: (Unnamed Layer* 0) [Convolution]
TENSORRT LOG: Internal error: could not find any implementation for node (Unnamed Layer* 0) [Convolution], try increasing the workspace size with IBuilder::setMaxWorkspaceSize()
TENSORRT LOG: ../builder/tacticOptimizer.cpp (1523) - OutOfMemory Error in computeCosts: 0
cloud not build cuda engine
/home/cecco/Documents/nn/tkDNN/src/NetworkRT.cpp:145
Aborting...
```

rod-hendricks, 2020-06-17:
@ceccocats Thanks for the reply. I ran it again in debug mode and apparently its a faulty installation of cublas on cuda. Im reinstalling all my prerequisite apps and specifically point to cuda 10.0 and tensorrt 6.0 and cudnn 7.6.5. I have some issues downloading a new tensorrt version as of the moment so I can't confirm yet if reinstalling everything will fix the issue.

rod-hendricks, 2020-06-18:
@ceccocats managed to clean up my installation and indeed the issue got fixed. However I ran into another problem which seems to point that i am missing weights for Convlution 488. I basically just ran ./test_yolo4 still and it automatically downloaded the binary files for yolov4. Any thoughts on what I might be missing?

[image]

rod-hendricks, 2020-06-18:
> @ceccocats managed to clean up my installation and indeed the issue got fixed. However I ran into another problem which seems to point that i am missing weights for my network. I basically just ran ./test_yolo4 still and it automatically downloaded the binary files for yolov4. Any thoughts on what I might be missing? I am running into this error on Layer 0 which is the first Conv2d layer.
> 
> [image]

Sorry I traced the error and figured out that I was missing TKDNN_MODE setting. Because it was null then the network did not have a precision setting. 

Now Im running into another error as shown below. Any Idea what might be causing this?

[image]

ceccocats, 2020-06-18:
I don't know, you have a lot of strange issues.
Have you tried to reboot? I know it's stupid but sometimes cuda go crazy.
Also delete all the build directory and rebuild everything.

rod-hendricks, 2020-06-19:
Finally solved it and it was my own doing again. I was deallocating memory after I create the network as I was ensuring that my prior issues were not memory related. I forgot to remove it which caused issues later on down the test line when the test script wanted to compare the results between the CuDNN network and TensorRT. After I removed the deallocation code everything works  perfect now. 

Thank you for all the help @ceccocats and apologies for the semi-wild goose chase.

closed 2020-09-11 by mive93