ceccocats/tkDNN #45: Error when caculate map

buiduchanh, 2020-06-18:
@ceccocats 
Hello, I have a small problem. I train yolov4 with my dataset (13 class), I prepare test data follow this step : 
- 2 foloder ./images and ./labels. Each .txt in labels has seperate line which follow format : 
`label x_center_box/img_width y_center_box/img_height width_box/img_width height_box/img_height `
- New config file like below

[image]

but when I run ./map_demo I have a problem 

[image]

Can you help me ? Thanks

closed 2020-06-18 by buiduchanh

mive93, 2020-06-18:
Hi @buiduchanh,

did you solve it?

buiduchanh, 2020-06-22:
@mive93 Sorry for late reply, I reslove this problem. This error comes from function for generate json result. I just turn off this flag and finish it