ceccocats/tkDNN #46: Error reading layer bin

Kr4is, 2020-06-18:
Hi @ceccocats !

I have the follow configuration file: 
[configuration.cfg.zip](https://github.com/ceccocats/tkDNN/files/4797737/configuration.cfg.zip)

When i export using the description in the readme it seems to be successfully:

network input size: 2076672
Predicted in 5.818229 seconds.

networks output size: 18928

I create a test, based on the yolo test, changing the yolo layers in the output_bins like this: 

``` cpp
#include<iostream>
#include<vector>
#include "tkdnn.h"
#include "test.h"
#include "DarknetParser.h"

int main() {
    std::string bin_path  = "configuration";
    std::vector<std::string> input_bins = { 
        bin_path + "/layers/input.bin"
    };
    std::vector<std::string> output_bins = {
        bin_path + "/debug/layer60_out.bin",
        bin_path + "/debug/layer66_out.bin",
        bin_path + "/debug/layer71_out.bin"
    };
    std::string wgs_path  = bin_path + "/layers";
    std::string cfg_path  = std::string(TKDNN_PATH) + "/tests/darknet/cfg/configuration.cfg";
    std::string name_path = std::string(TKDNN_PATH) + "/tests/darknet/names/configuration.names";
    // downloadWeightsifDoNotExist(input_bins[0], bin_path, "https://cloud.hipert.unimore.it/s/d97CFzYqCPCp5Hg/download");

    // parse darknet network
    tk::dnn::Network *net = tk::dnn::darknetParser(cfg_path, wgs_path, name_path);
    net->print();

    //convert network to tensorRT
    tk::dnn::NetworkRT *netRT = new tk::dnn::NetworkRT(net, net->getNetworkRTName(bin_path.c_str()));
    
    int ret = testInference(input_bins, output_bins, net, netRT);
    net->releaseLayers();
    delete net;
    delete netRT;
    return ret;
}
```
All compile successfully, but when i execute the test i obtain the follow output:

``` bash
./test_configuration
Not supported field: batch=128
Not supported field: subdivisions=32
Not supported field: momentum=0.9
Not supported field: decay=0.0005
Not supported field: angle=0
Not supported field: saturation = 1.5
Not supported field: exposure = 1.5
Not supported field: hue=.1
Not supported field: mixup=1
Not supported field: learning_rate=0.001
Not supported field: burn_in=9000
Not supported field: max_batches = 200000
Not supported field: policy=sgdr
Not supported field: sgdr_cycle=1000
Not supported field: sgdr_mult=2
Not supported field: steps=4000,6000,8000,9000
New NETWORK (tkDNN v0.5, CUDNN v7.605)
Reading weights: I=3 O=16 KERNEL=3x3x1
Reading weights: I=16 O=32 KERNEL=3x3x1
Not supported field: antialiasing=1
Reading weights: I=32 O=64 KERNEL=3x3x1
Not supported field: antialiasing=1
Reading weights: I=64 O=128 KERNEL=3x3x1
Not supported field: antialiasing=1
Reading weights: I=128 O=256 KERNEL=3x3x1
Not supported field: antialiasing=1
Reading weights: I=256 O=512 KERNEL=3x3x1
Reading weights: I=512 O=1024 KERNEL=3x3x1
Reading weights: I=1024 O=256 KERNEL=1x1x1
Not supported field: assisted_excitation=4000
Reading weights: I=256 O=512 KERNEL=3x3x1
Reading weights: I=512 O=128 KERNEL=1x1x1
Reading weights: I=384 O=128 KERNEL=1x1x1
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=256 O=64 KERNEL=1x1x1
Reading weights: I=64 O=128 KERNEL=3x3x1
Reading weights: I=32 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=256 O=64 KERNEL=1x1x1
Reading weights: I=512 O=64 KERNEL=1x1x1
Not supported field: maxpool_depth=1
Reading weights: I=768 O=128 KERNEL=3x3x1
Error reading file configuration/layers/c58.bin with n of float: 884736 seek: 0 size: 3538944

/home/mads/bruno/Documentos/tkDNN/src/utils.cpp:58
Aborting...
```

What can be caused this? Thank you in advance!

ceccocats, 2020-06-18:
Hi, when it prints:
`Not supported field ...`
It means that the field in the cfg is not taken into account changing the model of the network and therefore the dimension of the weights to load.
Not all cfg are supported, see DarknetParser.cpp

closed 2020-06-23 by Kr4is

ASONG0506, 2020-06-23:
Hi, I met the similar problem.
I trained my yolov4 model using my own dataset with 10 class, the I converted it using the [this repo](https://git.hipert.unimore.it/fgatti/darknet) and no error occured with these infomation:
```
......

n: 136, type 0
Convolutional
weights: 32768, biases: 128, batch_normalize: 1, groups: 1
write binary ../output_bdd100k/c136.bin

n: 137, type 0
Convolutional
weights: 294912, biases: 256, batch_normalize: 1, groups: 1
write binary ../output_bdd100k/c137.bin

n: 138, type 0
Convolutional
weights: 11520, biases: 45, batch_normalize: 0, groups: 1
write binary ../output_bdd100k/c138.bin

n: 139, type 27
export YOLO
mask: 3
biases: 18
mask 0.000000
mask 1.000000
mask 2.000000
anchor 12.000000
anchor 16.000000
anchor 19.000000
anchor 36.000000
anchor 40.000000
anchor 28.000000
anchor 36.000000
anchor 75.000000
anchor 76.000000
anchor 55.000000
anchor 72.000000
anchor 146.000000
anchor 142.000000
anchor 110.000000
anchor 192.000000
anchor 243.000000
anchor 459.000000
anchor 401.000000
write binary ../output_bdd100k/g139.bin

n: 140, type 9
export ROUTE

n: 141, type 0
Convolutional
weights: 294912, biases: 256, batch_normalize: 1, groups: 1
write binary ../output_bdd100k/c141.bin

n: 142, type 9
export ROUTE

n: 143, type 0
Convolutional
weights: 131072, biases: 256, batch_normalize: 1, groups: 1
write binary ../output_bdd100k/c143.bin

n: 144, type 0
Convolutional
weights: 1179648, biases: 512, batch_normalize: 1, groups: 1
write binary ../output_bdd100k/c144.bin

n: 145, type 0
Convolutional
[... 323 lines cut ...]
random=1
scale_x_y = 1.05
iou_thresh=0.213
cls_normalizer=1.0
iou_normalizer=0.07
iou_loss=ciou
nms_kind=greedynms
beta_nms=0.6
max_delta=5

```

Then I use this repo to generate the .rt file and do inference, when run this commond `./test_yolo4` and error occured as follow:
```
.......
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=255 KERNEL=1x1x1
Error reading file yolo4/layers/c138.bin with n of float: 65280 seek: 0 size: 261120

/home/xavier/test_ws/tkDNN-master/src/utils.cpp:58
Aborting...

```

Do you know what's whe wrong with it, thanks!

ASONG0506, 2020-06-23:
> Hi, I met the similar problem.
> I trained my yolov4 model using my own dataset with 10 class, the I converted it using the [this repo](https://git.hipert.unimore.it/fgatti/darknet) and no error [...]

I found the solution, thanks

ceccocats, 2020-06-25:
Hi, can you share your solution? It can be useful.
It was your mistake? or an error in the code?

ceccocats, 2020-06-25:
Yes, darknet Is big and the time is limited.
We decided to implement only the official YOLOs and the one that needs only minor changes.

ASONG0506, 2020-06-25:
There is no error in your code, actually I just changed my own cfg file and the default cfg file for parsing the darknet network is : `root/tests/darknet/cfg/yolov4.cfg` and the corresponding names file is : `root/tests/darknet/names/coco.names`. when I modified these two files, it worked!

zjZSTU, 2020-07-30:
> Hi, when it prints:
> `Not supported field ...`
> It means that the field in the cfg is not taken into account changing the model of the network and therefore the dimension of the weights to load.
> Not all cfg are supported, see DarknetParser.cpp

hi @ceccocats , i use yolov4 pretrained weights in [YOLOv4 model zoo](https://github.com/AlexeyAB/darknet/wiki/YOLOv4-model-zoo), there also will appear `Not supported` info, but following work is fine

```
$ ./test_yolo4
Not supported field: batch=1
Not supported field: subdivisions=1
Not supported field: momentum=0.949
Not supported field: decay=0.0005
Not supported field: angle=0
Not supported field: saturation = 1.5
Not supported field: exposure = 1.5
Not supported field: hue=.1
Not supported field: learning_rate=0.00261
Not supported field: burn_in=1000
Not supported field: max_batches = 500500
Not supported field: policy=steps
Not supported field: steps=400000,450000
Not supported field: scales=.1,.1
Not supported field: mosaic=1
New NETWORK (tkDNN v0.5, CUDNN v8)
```

but when i use my own trained weights,  it doesn't work

```
$ ./test_yolo4
Not supported field: batch=1
Not supported field: subdivisions=1
Not supported field: momentum=0.949
Not supported field: decay=0.0005
Not supported field: angle=0
Not supported field: saturation = 1.5
Not supported field: exposure = 1.5
Not supported field: hue=.1
Not supported field: learning_rate=0.00261
Not supported field: burn_in=1000
Not supported field: max_batches = 500500
Not supported field: policy=steps
Not supported field: steps=400000,450000
Not supported field: scales=.1,.1
Not supported field: mosaic=1
New NETWORK (tkDNN v0.5, CUDNN v8)
Reading weights: I=3 O=32 KERNEL=3x3x1
Reading weights: I=32 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=32 KERNEL=1x1x1
Reading weights: I=32 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=64 O=128 KERNEL=3x3x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=64 O=64 KERNEL=3x3x1
Reading weights: I=64 O=64 KERNEL=1x1x1
Reading weights: I=128 O=128 KERNEL=1x1x1
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=256 O=128 KERNEL=1x1x1
...
...
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=255 KERNEL=1x1x1
Error reading file yolo4/layers/c138.bin with n of float: 65280 seek: 0 size: 261120

/home/user/software/tkDNN/src/utils.cpp:58
Aborting...
```

thank you for you help

mive93, 2020-08-06:
Hi @zjZSTU 

which cfg are you using in the test?

zjZSTU, 2020-08-06:
> Hi @zjZSTU
> 
> which cfg are you using in the test?

hi @mive93 , i have solved my problem. the key is to choose my own .cfg and .name file for custom dataset. refer to #99

mive93, 2020-08-06:
Yes, exactly :+1: