ceccocats/tkDNN #50: Why does yolov3 run slower on Tesla T4 than on RTX2080ti？

rotorliu, 2020-06-21:
GeForce RTX 2080 Ti：FP16 (half) performance 26.90 TFLOPS (2:1)
Tesla T4：FP16 (half) performance 65.13 TFLOPS (8:1)

**Images Per Second**
 BatchSize | T4 | RTX2080ti
-- | -- | --
1 | 145.507 | 197.528
2 | 241.571 | 286.453
4 | 260.832 | 430.495
8 | 265.315 | 520.995
16 | 283.619 | 488.933
32 | 256.451 | 550.153
64 | 245.872 | 554.456
128 | 243.505 | 547.426
256 | 228.099 | ——

PS：Testing with tkDNN

mrhosseini, 2020-06-21:
Actually RTX2080ti is faster than Tesla T4. See the below links:

https://www.servethehome.com/nvidia-tesla-t4-ai-inferencing-gpu-review/4/

https://versus.com/en/nvidia-geforce-rtx-2080-ti-founders-edition-vs-nvidia-tesla-t4

rotorliu, 2020-06-21:
@mrhosseini Thank you
 Why is it that T4 FLOPs is higher but slower？

mrhosseini, 2020-06-21:
> Why is it that T4 FLOPs is higher but slower？
On FP32 RTX2080ti is faster. On FP16 T4 has a higher FLOPS and should be faster. If not may be possible bug in tkDNN or TensorRT. May be not you are not using the latest version TensorRT.

rotorliu, 2020-06-21:
> > Why is it that T4 FLOPs is higher but slower？
> 
> On FP32 RTX2080ti is faster. On FP16 T4 has a higher FLOPS and should be faster. If not may be possible bug in tkDNN or TensorRT. May be not you are not using the latest version TensorRT.

The table above is the result of FP16 mode. The TensorRT version is v7. So I am confused.
The code is used：
https://github.com/ceccocats/tkDNN/blob/master/tests/test_rtinference/rtinference.cpp

rotorliu, 2020-06-21:
[image]
This test also shows that T4 is relatively slow.

ceccocats, 2020-06-21:
[image]

[image]

Source:
https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.nvidia.com/content/dam/en-zz/Solutions/design-visualization/technologies/turing-architecture/NVIDIA-Turing-Architecture-Whitepaper.pdf&ved=2ahUKEwjwsLzNtZPqAhXC3KQKHUJfCLUQFjAGegQIARAB&usg=AOvVaw3CuG9m9TLCpXjybLqkvyJR&cshid=1592760127145

closed 2020-06-30 by rotorliu