ceccocats/tkDNN #52: How to get the yolo4_fp16.rt file of  yolov4.cfg trained on my data?

dadaligoudan, 2020-06-22:
I want to get the .rt file of my own model (cfg and weights file), how can i solve this? Thanks a lot.

mive93, 2020-06-24:
You have to export the weights (https://github.com/ceccocats/tkDNN#1export-weights-from-darknet) and create a test for your cfg similar to [this](https://github.com/ceccocats/tkDNN/blob/master/tests/darknet/yolo4.cpp). Remember to change also the classes names.

dadaligoudan, 2020-06-24:
> You have to export the weights (https://github.com/ceccocats/tkDNN#1export-weights-from-darknet) and create a test for your cfg similar to [this](https://github.com/ceccocats/tkDNN/blob/master/tests/darknet/yolo4.cpp). Remember to change also the classes names.

Thanks a lot, I have finished this. I have some other questions. During the INT8 inference, do i have to offer 100 images to create the calibration table? Moreover, Do the labels needed is the same format as the AlexeyAB project needed, which is  <object-class> <x_center> <y_center> <width> <height>? appreciate for your reply.

dadaligoudan, 2020-06-24:
Thanks a lot, I have finished this. I have some other questions. During the INT8 inference, do i have to offer 100 images to create the calibration table? Moreover, Do the labels needed is the same format as the AlexeyAB project needed, which is  <object-class> <x_center> <y_center> <width> <height>?  Appreciate for your reply.

dadaligoudan, 2020-06-24:
> You have to export the weights (https://github.com/ceccocats/tkDNN#1export-weights-from-darknet) and create a test for your cfg similar to [this](https://github.com/ceccocats/tkDNN/blob/master/tests/darknet/yolo4.cpp). Remember to change also the classes names.

Thanks a lot, I have finished this. I have some other questions. During the INT8 inference, do i have to offer 100 images to create the calibration table? Moreover, Do the labels needed is the same format as the AlexeyAB project needed, which is  <object-class> <x_center> <y_center> <width> <height>?  Appreciate for your reply.

dadaligoudan, 2020-06-24:
> You have to export the weights (https://github.com/ceccocats/tkDNN#1export-weights-from-darknet) and create a test for your cfg similar to [this](https://github.com/ceccocats/tkDNN/blob/master/tests/darknet/yolo4.cpp). Remember to change also the classes names.

Thanks a lot, I have finished this. I have some other questions. During the INT8 inference, do i have to offer 100 images to create the calibration table? Moreover, Do the labels needed is the same format as the AlexeyAB project needed, which is  object-class x_center y_center width and height?  Appreciate for your reply.

mive93, 2020-06-24:
Hi @dadaligoudan 
Yes, the labels should be in that format. 
And yesh, 100 images are needed, but you can also increase that number if you want. With more images you should obtain a better calibration.

alphapibeta, 2020-07-22:
For those who are new to c++ and dont know how to generate a .trt engine for custom objects.

**Step 1**
Clone the repo [darknet](https://github.com/ceccocats/tkDNN#1export-weights-from-darknet) follow the instructions and generate two directories 
**debug** and **layers** (they contain weights)

**Step 2**
Select your architecture from the list 
```
tkDNN/tests/darknet/
```
and make a copy of the yolo4tiny.cpp. Example:**yolo4tiny_custom.cpp**

**Step 3**
Create a directory for your model in the build folder; for example yolo4tiny_custom
dump **layers** and **debug** generated from **Step 1** into **yolo4tiny_custom** directory
update the [binpath](https://github.com/ceccocats/tkDNN/blob/f4970d1e6faab505c2caf1d6833cf7490a971a0e/tests/darknet/yolo4tiny.cpp#L8) string with new directory **yolo4tiny_custom** in your **yolo4tiny_custom.cpp**

Also, add **yolo4tiny_custom.cfg** and **yolo4tiny_custom.names** to
```
/tkDNN/tests/darknet/cfg/
/tkDNN/tests/darknet/names/
```
respectively.
Replace the existing path with new file path [yolo4tiny_custom.cfg](https://github.com/ceccocats/tkDNN/blob/f4970d1e6faab505c2caf1d6833cf7490a971a0e/tests/darknet/yolo4tiny.cpp#L17), [yolo4tiny_custom.names](https://github.com/ceccocats/tkDNN/blob/f4970d1e6faab505c2caf1d6833cf7490a971a0e/tests/darknet/yolo4tiny.cpp#L18)

**Step 4**
run from the build
```
$camke ..
$make
```
This shall generate a new executable file **yolo4tiny_custom**
run 
```
./yolo4tiny_custom
```
this will generate **yolo4tiny_custom.trt**

maohu-96, 2020-07-28:
Hi！
After make is completed, where can I find the executable file, as you said yolov4tiny_custom

maohu-96, 2020-07-28:
> 对于那些不熟悉c ++并且不知道如何为自定义对象生成.trt引擎的用户。
> 
> **步骤1**按照说明
> 克隆repo [darknet](https://github.com/ceccocats/tkDNN#1export-weights-from-darknet)，并生成两个目录
> **debug**和**layer**（它们包含权重）
> 
> **步骤2**
> 从列表中选择您的架构
> 
> ```
> tkDNN/tests/darknet/
> ```
> 
> 并复制yolo4tiny.cpp。示例：**yolo4tiny_custom.cpp**
> 
> **步骤3**
> 在build文件夹中为您的模型创建一个目录；例如yolo4tiny_custom
> 转储**层**和**调试**，从产生的**第1步**到**yolo4tiny_custom**目录
> 更新[binpath](https://github.com/ceccocats/tkDNN/blob/f4970d1e6faab505c2caf1d6833cf7490a971a0e/tests/darknet/yolo4tiny.cpp#L8)串新目录**yolo4tiny_custom**在**yolo4tiny_custom.cpp**
> 
> 另外，将**yolo4tiny_custom.cfg**和**yolo4tiny_custom.names**添加到
> 
> ```
> /tkDNN/tests/darknet/cfg/
> /tkDNN/tests/darknet/names/
> ```
> 
> 分别。
> 用新文件路径[yolo4tiny_custom.cfg](https://github.com/ceccocats/tkDNN/blob/f4970d1e6faab505c2caf1d6833cf7490a971a0e/tests/darknet/yolo4tiny.cpp#L17)，[yolo4tiny_custom.names](https://github.com/ceccocats/tkDNN/blob/f4970d1e6faab505c2caf1d6833cf7490a971a0e/tests/darknet/yolo4tiny.cpp#L18)替换现有路径[](https://github.com/ceccocats/tkDNN/blob/f4970d1e6faab505c2caf1d6833cf7490a971a0e/tests/darknet/yolo4tiny.cpp#L18)
> 
> **步骤4：**
> 从构建开始
> 
> ```
> $camke ..
> $make
> ```
> 
> 这将生成一个新的可执行文件**yolo4tiny_custom**
> run
> 
> ```
> ./yolo4tiny_custom
> ```
> 
> 这将生成**yolo4tiny_custom.trt**

Hi！
After make is completed, where can I find the executable file, as you said yolov4tiny_custom

alphapibeta, 2020-07-28:
> Hi！
> After make is completed, where can I find the executable file, as you said yolov4tiny_custom

@maohu-96 

you find the **yolov4tiny_custom** in the **build** directory after executing these commands
```
/path/to/tkdnn/$ mkdir build ; cd build
build $ cmake ..
build $ make
```

maohu-96, 2020-07-29:
> > > 对于那些不熟悉C ++并且不知道如何为自定义对象生成.TRT引擎的用户。
> > > **步骤1**按照说明
> > > 克隆回购[暗网](https://github.com/ceccocats/tkDNN#1export-weights-from-darknet)，并生成两个目录
> > > **调试**和**层**（它们包含权重）
> > > **步骤2** 
> > > 从列表中选择您的架构
> > > ```
> > > tkDNN/tests/darknet/
> > > ```
> > > 
> > > 
> > > 并复制yolo4tiny.cpp示例：**yolo4tiny_custom.cpp **
> > > **步骤3** 
> > > 在生成文件夹中为您的模型创建一个目录;例如yolo4tiny_custom 
> > > 转储**层**状语从句：**调试**，产生从的**第1步**到**yolo4tiny_custom**目录
> > > 更新[binpath](https://github.com/ceccocats/tkDNN/blob/f4970d1e6faab505c2caf1d6833cf7490a971a0e/tests/darknet/yolo4tiny.cpp#L8)串新目录**yolo4tiny_custom**在**yolo4tiny_custom.cpp另外** 
> > > ，将**yolo4tiny_custom.cfg**和**yolo4tiny_custom.names**添加到
> > > ```
> > > /tkDNN/tests/darknet/cfg/
> > > /tkDNN/tests/darknet/names/
> > > ```
> > > 
> > > 
> > > 分别。
> > > 用新文件路径[yolo4tiny_custom.cfg](https://github.com/ceccocats/tkDNN/blob/f4970d1e6faab505c2caf1d6833cf7490a971a0e/tests/darknet/yolo4tiny.cpp#L17)，[yolo4tiny_custom.names替换](https://github.com/ceccocats/tkDNN/blob/f4970d1e6faab505c2caf1d6833cf7490a971a0e/tests/darknet/yolo4tiny.cpp#L18)现有路径
> > > **步骤4：** 
> > > 从生成开始
> > > ```
> > > $camke ..
> > > $make
> > > ```
> > > 
> > > 
> > > 这将生成一个新的重置文件**yolo4tiny_custom** 
> > > run
> > > ```
> > > ./yolo4tiny_custom
> > > ```
> > > 
> > > 
> > > 这将生成**yolo4tiny_custom.trt**
> > 
> > 
> > 嗨！
> > make完成后，您可以在哪里找到可执行文件，如您所说yolov4tiny_custom
> 
> @ maohu-96
> 
> 执行以下命令后，您会在**构建**目录中找到**yolov4tiny_custom******
> 
> ```
> /path/to/tkdnn/$ mkdir build ; cd build
> build $ cmake ..
> build $ make
> ```

Thank you for your prompt reply！！
I make it again according to your method, but the following error occurred after make
[image]
How can i solve it

alphapibeta, 2020-07-29:
> Thank you for your prompt reply！！
> I make it again according to your method, but the following error occurred after make
> [image]
> How can i solve it

@maohu-96 

Please install this dependency 

```
sudo apt-get install libyaml-cpp-dev
```
or 
```
sudo apt-get install libyaml-cpp0.3-dev
```

maohu-96, 2020-07-29:
> > > > > 对于那些不熟悉C ++并且不知道如何为自定义对象生成.TRT引擎的用户。
> > > > > **步骤1**按照说明
> > > > > 克隆回购[暗网](https://github.com/ceccocats/tkDNN#1export-weights-from-darknet)，生成并目录两个
> > > > > **调试**状语从句：**层**（它们包含权重）
> > > > > **步骤2** 
> > > > > 从列表中选择您的架构
> > > > > ```
> > > > > tkDNN/tests/darknet/
> > > > > ```
> > > > > 
> > > > > 
> > > > > 并复制yolo4tiny.cpp示例：**yolo4tiny_custom.cpp ** 
> > > > > **步骤3** 
> > > > > 在生成文件夹中为您的模型创建一个目录;例如yolo4tiny_custom 
> > > > > 转储**层**状语从句：**调试**，从产生的**第1步**到** yolo4tiny_custom **目录
> > > > > 更新[binpath](https://github.com/ceccocats/tkDNN/blob/f4970d1e6faab505c2caf1d6833cf7490a971a0e/tests/darknet/yolo4tiny.cpp#L8)串新目录** yolo4tiny_custom **在** yolo4tiny_custom.cpp另外** 
> > > > > ，将**yolo4tiny_custom.cfg**和**yolo4tiny_custom.names**添加到
> > > > > ```
> > > > > /tkDNN/tests/darknet/cfg/
> > > > > /tkDNN/tests/darknet/names/
> > > > > ```
> > > > > 
> > > > > 
> > > > > 分别。
> > > > > 用新文件路径[yolo4tiny_custom.cfg](https://github.com/ceccocats/tkDNN/blob/f4970d1e6faab505c2caf1d6833cf7490a971a0e/tests/darknet/yolo4tiny.cpp#L17)，[yolo4tiny_custom.names替换](https://github.com/ceccocats/tkDNN/blob/f4970d1e6faab505c2caf1d6833cf7490a971a0e/tests/darknet/yolo4tiny.cpp#L18)现有路径
> > > > > **步骤4：** 
> > > > > 从生成开始
> > > > > ```
> > > > > $camke ..
> > > > > $make
> > > > > ```
> > > > > 
> > > > > 
> > > > > 这将生成一个新的重置文件**yolo4tiny_custom** 
> > > > > run
> > > > > ```
> > > > > ./yolo4tiny_custom
> > > > > ```
> > > > > 
> > > > > 
> > > > > 这将生成**yolo4tiny_custom.trt**
> > > > 
> > > > 
> > > > 嗨！
> > > > make完成后，您可以在哪里找到重组文件，如您所说的yolov4tiny_custom
> > > 
> > > 
> > > @ maohu-96 
> > > 执行以下命令后，您会在**构建**目录中找到**yolov4tiny_custom** ****
> > > ```
> > > /path/to/tkdnn/$ mkdir build ; cd build
> > > build $ cmake ..
> > > build $ make
> > > ```
> > 
> > 
> > 感谢您的迅速答复！！
> > 我根据您的方法再次进行制作，但是make之后出现以下错误
> > 我该如何解决
> > [image]
> 
> @ maohu-96
> 
> 请安装此依赖项
> 
> ```
> sudo apt-get install libyaml-cpp-dev
> ```
> 
> 要么
> 
> ```
> sudo apt-get install libyaml-cpp0.3-dev
> ```

Thank you for your prompt reply！！
I installed yaml-cpp, and the following error occurred after make，how can i solve it?

[image]

alphapibeta, 2020-07-30:
@

> Thank you for your prompt reply！！
> I installed yaml-cpp, and the following error occurred after make，how can i solve it?
>
> [image]

@maohu-96 

This is a CMake issue, the below command should be able to fix it. I did not face this issue while building.
```
sudo apt-get install libyaml-cpp-dev
```

mive93, 2020-09-11:
Closing for now, 
feel free to reopen.

closed 2020-09-11 by mive93

Goru1890, 2020-11-25:
> This is a CMake issue, the below command should be able to fix it. I did not face this issue while building.
> 
> ```
> sudo apt-get install libyaml-cpp-dev

Didn't work for me.
> ```