ceccocats/tkDNN #56: Recorded FPS vs actual FPS

sambo55, 2020-06-23:
I'm struggling to achieve the FPS reported in the command line.

For example when I run inference on a 10 min 30fps video the reported inference fps is 300+.

I would expect that the time taken to run inference on the entire video would be 30x60x10 = 18000 / 300 fps = 60secs = 1min

Yet the code takes at least 3 mins to run. Is there something wrong with my calculation? Why would the reported fps not be the actual?

ceccocats, 2020-06-23:
Hi sambo,
"Inference time" is only inference time, it doesn't count preprocessing, postprocessing and visualisation

sambo55, 2020-06-24:
Thanks. Any pointers on how to optimise those aspects?

ceccocats, 2020-06-24:
Opencv Is comfortable but slow expecially for visualization. Find an opengl viewer that fit your needs.
For preprocessing and postprocessing ensure that your are compiling opencv with CUDA and cudacodec

mive93, 2020-06-29:
Hi @sambo55,

If you have opencv (4.x) with contrib compiled for CUDA, then you can uncomment line 17 [here](https://github.com/ceccocats/tkDNN/blob/master/include/tkDNN/DetectionNN.h). The preprocessing will be optimized on GPU. 
Another thing you could do is decouple inference and visualization, using different threads.

rod-hendricks, 2020-07-03:
I tried running Batchsize=1 vs 4 and I notice that the inference speed is not as fast as I thought it would be on a RTX2070. For size=1 I get ~6.6ms inf time while when i do size=4 I get ~17.7ms (2.6+ slower). Are these numbers correct? I am only using ~1.6Gb of GPU mem and about ~40% processing power even when running size 4. 

Would you guys know of a way to optimize this by utilizing more of the GPU power? 

On a side note, my pre and post processing times are awful when I do size=4 which are running a total of ~9ms. I will try to check opencv again and ensure compiled on CUDA to see if it significantly improves.

rod-hendricks, 2020-07-13:
I built opencv-4.2.0 with cuda and cudnn enabled and found that there was no substantial improvement gained on the pre and post processing portion of Yolo3Detection.cpp. In fact, it was slower in my end if i enable (uncomment https://github.com/ceccocats/tkDNN/blob/7c2155decfc2d225f523350d55fb6d773d3b3b6c/include/tkDNN/DetectionNN.h#L17) OPENCV_CUDACONTRIB as compared to running it disabled. Am I doing this right? 

Also I guess the results I have on inference speed cannot be optimized anymore on my hardware?

rod-hendricks, 2020-07-16:
Thanks for the response and advice @mive93 ! I am not sure yet as to the effort vs gain on doing this so I'll see if I can consider working on this later. 

I would like to ask further though about what you meant by Yolo having too many useless passages between host and device. Do you mean that within the yolo inference task, there is still the data passages going on between host and device before the network output is produced?

mive93, 2020-07-16:
@rod-hendricks if I have updates on any improvements, I will let you know.

No, I meant only on our preprocessing. The code should be cleaned up and fix to remove a useless passage between host and device. When I'll have time I'll fix that :)

rod-hendricks, 2020-07-16:
Thanks @mive93 ! Much appreciated. Good work on this repo. Its amazing!

closed 2020-09-11 by mive93

MohammadKassemZein, 2021-08-13:
Hello @mive93 , any updates on removing the useless passage between host and device?
Thanks.

mive93, 2021-08-19:
HI @MohammadKassemZein,
no updates yet, I'm sorry.

mkzein, 2021-11-09:
@mive93 Sorry to bother :) any updates on this?

mive93, 2022-01-19:
Not yet, but maybe soon. 
We have already the code in an internal project, we just need to merge it here.

mkzein, 2022-01-20:
Sounds great!
Thank you @mive93

mive93, 2022-01-26:
[image] [Fix tkDNN preprocessing](https://trello.com/c/cUcuGEC5/357-fix-tkdnn-preprocessing)