ceccocats/tkDNN #57: INT8 inference can not finish correctly

dadaligoudan, 2020-06-24:
I have some questions about INT8 inference. During the INT8 inference, do i have to offer 100 images to create the calibration table? Moreover, Do the labels needed is the same format as the AlexeyAB project needed, which is object-class x_center y_center width and height? I have the image_list.txt and the label_list.txt followed by your instruction, but it doesn't work. Much appreciate for your reply.

mive93, 2020-06-24:
HI @dadaligoudan,
already replied in the other issue.

What does it mean that it does not work?
Can you copy here your error please.

omaralvarez, 2020-06-25:
I think my pull request can fix the issue. Sounds like what happenned to me, error when parsing classes #53.

dadaligoudan, 2020-06-28:
> HI @dadaligoudan,
> already replied in the other issue.
> 
> What does it mean that it does not work?
> Can you copy here your error please.

Hi， @mive93 thanks for your reply.  @omaralvarez has the same problem with me.It is because it has a black space before \n and it goes to a infinite loop. But i got another problem as follows:  It has finished 99 batches,but it stopped during the 99 batch, seems it went to another infinite loop. I have checked this picture and the corresponding label and it has no problem with it. Do you know why? By the way, the pictures i used is 1280*1024 and the label file is generated by this resolution.
[image]

dadaligoudan, 2020-06-28:
> HI @dadaligoudan,
> already replied in the other issue.
> 
> What does it mean that it does not work?
> Can you copy here your error please.

Hi, @mive93 Thanks for your reply, I seems i have the same problem with @omaralvarez ,this is because in each line of the label file, it has a black space before \n, and it goes to an  infinite loop. I have fixed this problem, but another problem occurred. After INT8 inference, the network can't detect any objects, seems something goes wrong. Do you know why? By the way, the pictures i used is 1280*1024 and the labels is generated according to that resolution.

omaralvarez, 2020-06-28:
I somehow had the same issues. My optimized YOLO version did not function after conversion to INT8. It may be due to the images chosen for calibration also, the images chosen have to be representative of the cases that the network will see later on.

mive93, 2020-06-29:
Hi @dadaligoudan,

yes, there was a bug that has been fixed, on the \n.
For the quantization, @omaralvarez is right. It could be that you have picked wrong images, or not enough (try maybe to increase that 100 to 1000). An high accuracy drop with int8 is very common. 
Or is it that you do not have any boxes? That would be weird.

mive93, 2020-09-11:
Closing for inactivity, 
feel free to reopen.

closed 2020-09-11 by mive93

MohamedElsaeidy, 2021-09-05:
> Hi @dadaligoudan,
>
> yes, there was a bug that has been fixed, on the \n.
> For the quantization, @omaralvarez is right. It could be that you have picked wrong images, or not enough (try maybe to [...]

how we can increase that 100 image to 1000 for calibration ? in which file ?

mive93, 2021-09-06:
@MohamedElsaeidy here: https://github.com/ceccocats/tkDNN/blob/master/src/NetworkRT.cpp#L85