ceccocats/tkDNN #58: Error with new yolov4 config

buiduchanh, 2020-06-25:
Hello. I have a small question. I check update of @alexeyAB for new yolov4 with mish activation, this is the new config 
https://drive.google.com/file/d/1V726wzWW1iBDcJ7uP8c0n_Y4M98lIg-C/view

I train the model with this config but when It can not export to `layers` and `debug`. The error I mentioned below 
[image]

This is the layer 137, Alexab add new layer "scale Layer" and I think this is the main reason

[image]

Can you share your solution for fix it ? Thanks

ceccocats, 2020-06-25:
Hi,
Where is? I can't find it in Alexey repo.
If it has a better map we will implement it for sure.

buiduchanh, 2020-06-25:
@ceccocats  
This is in yolov4 model zoo : YOLOv4-SAM-Mish
https://github.com/AlexeyAB/darknet/wiki/YOLOv4-model-zoo
It add Sam layer so may be cause bug

ceccocats, 2020-06-25:
Yes is a new layer.
Looking at the code it seems a shortcut with multiplication instead of sum and an activation.
We already have the shortcut with mul in the branch segmentation. Maybe next week we will merge in master.

closed 2020-06-30 by buiduchanh

Goru1890, 2020-11-12:
How much does YOLOv4-SAM-Mish improve detection over YOLOv4?

mive93, 2020-11-16:
Hi @Goru1890, that's more a question for the developers od yolov4.
However, [here](https://github.com/AlexeyAB/darknet/wiki/YOLOv4-model-zoo#yolov4-model-zoo) is the answer at your question:

[image]