ceccocats/tkDNN #60: Bug: incorrect output dimensions in yolo layer

mrhosseini, 2020-06-28:
In yolov3_tiny network dimensions of the first yolo layer is `13 x 13 x 255` and dimensions of the second one is `26 x 26 x 255`. However, if we use a network configuration in which unlike yolov3_tiny the dimensions of the first yolo layer is bigger than the second one ([like this](https://github.com/ceccocats/tkDNN/files/4776580/config.cfg.txt)),[ this line:](https://github.com/ceccocats/tkDNN/blob/6d9beb1ec5da217d8c6b1b2b51bfabf6766c3f95/src/Yolo.cpp#L123) 
```C
checkCuda( cudaMemcpy(predictions, dstData, output_dim.tot()*sizeof(dnnType), cudaMemcpyDeviceToHost));
```
will fail with the error:
```
Cuda failure: invalid argument
```

The problem is that `output_dim` of the layer is different from the dimensions of the `dstData` and with debugging it can be found that the for the first yolo layer the `output_dim` of it equals to the second one and vice versa. (Or we can say that the dimensions of  `dstData` belongs the other layer). 

I'm not sure if other parameters of the layer has the same problem.

mrhosseini, 2020-06-28:
After more debugging I found that, order of yolo layers in `pluginFactory` of `NetworkRT` after desrialization, is different from what we have in `NetworkRT::buffersRT` which is based on `engineRT->getBindingDimensions()`.

mrhosseini, 2020-06-28:
Changing [this line](https://github.com/ceccocats/tkDNN/blob/6d9beb1ec5da217d8c6b1b2b51bfabf6766c3f95/src/Yolo3Detection.cpp#L95) to the following will solve the problem. But I am not sure if this works for all the cases.

```C
rt_out[i] = (dnnType *) netRT->buffersRT[netRT->pluginFactory->n_yolos - 1] + netRT->buffersDIM[netRT->pluginFactory->n_yolos - 1].tot() * bi;
```

ceccocats, 2020-07-06:
Hi,
The order of buffersRt should always be:
Input
Output 0
Output 1
Etc.

If this really happen I need to check

mrhosseini, 2020-07-06:
> The order of buffersRt should always be:
> Input
> Output 0
> Output 1
> Etc.

What happens is that order of yolo layers in `NetworkRT::pluginFactory` is different from `bufferRT`. Example for 2 yolo layers :
```
Input
Output 0 [26 x 26 x 18]
Output 1 [13 x 13 x 18]
```
in `bufferRT` but in `pluginFactory`:
```
Yolo 0 [13 x 13 x 18]
Yolo 1 [26 x 26 x 18]
```