ceccocats/tkDNN #62: met error when complie

huaze555, 2020-06-29:
install dir:/usr/local
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_nvinfer_LIBRARY
    linked by target "tkDNN" in directory /home/zengxiaojia/CLionProjects/tkDNN

-- Configuring incomplete, errors occurred!

omaralvarez, 2020-06-29:
CMake is not able to find TensorRT, you probably have not installed TensorRT in the correct system paths. You should follow nvidia install instructions.

ceccocats, 2020-06-29:
Check your cmake version, =<3.15 is reccomended

huaze555, 2020-06-30:
@omaralvarez thanks for your quily reply. I am download the tar package of TensorRT. could I provide the include and lib path in the cmakelists.txt ?

tosonw, 2020-06-30:
@huaze555 你好，请问你这个问题解决了吗，我碰到了同样的问题，我更新cmake到了3.15.3，仍然没有得到解决

huaze555, 2020-06-30:
@tosonw 在cmakelists.txt里手动添加tensorrt的头文件和库文件路径。类似下面：

set(TensorRT_DIR "/home/***/TensorRT-6.0.1.5")

include_directories(${CMAKE_CURRENT_SOURCE_DIR}/include/tkDNN  ${TensorRT_DIR}/include)
link_directories($${TensorRT_DIR}/lib)     #  -L

tosonw, 2020-06-30:
@ceccocats 好的，我试试，谢谢

luckcodingdog, 2020-07-06:
@ceccocats @tosonw @huaze555  
I have encountered the same question.
My OS is centos, and use .tar.gz file to install the TensorRT.
I have update my cmake to 3.17.3.
Have you encountered the same question, everybody?
Thanks a lot in advance!

ceccocats, 2020-07-06:
You need to install CUDA, cuDNN and tensorrt.
The easiest way is to use NVIDIA jetpack and install all the HOST software.

luckcodingdog, 2020-07-07:
@ceccocats  thanks ！
I have try some times, but can't compile successfully.
I have  installed CUDA 10.0， cuDNN 7.5， TensorRT 7.0 and add lib as environment variable.
I install the TensorRT following the 4.3 part of  the guide of  https:i//docs.nvidia.com/deeplearning/tensorrt/install-guide/ .

ceccocats, 2020-07-14:
You can change 
```
  find_cuda_helper_libs(nvinfer)
  set(NVINFER_LIBRARY ${CUDA_nvinfer_LIBRARY} CACHE FILEPATH "location of the nvinfer library")
  unset(CUDA_nvinfer_LIBRARY CACHE)
```

with the loacation of libnvinfer.so in your PC: 
```
  set(NVINFER_LIBRARY "/path/of/libnvinfer.so")
```
or even with only the name, if the library is in LD_LIBRARY_PATH:
```
  set(NVINFER_LIBRARY nvinfer)
```

then do cmake and check the location of NVINFER inf the cmake output, you should have something like this:

luckcodingdog, 2020-07-21:
Thanks @ceccocats ,
I have solve this problem according to your suggestion.

basailuonaaa, 2020-07-31:
> Thanks @ceccocats ,
> I have solved this problem according to your suggestion.

hello,luckcodingdog,I met the same problem. I set(NVINFER_LIBRARY nvinfer),but another problem again,How do you solve this problem: 
-- Found CUDNN: /usr/local/cuda-10.0/lib64/libcudnn.so
-- Found CUDNN include: /usr/local/cuda-10.0/include
-- Found NVINFER: nvinfer
-- Found NVINFER include: NVINFER_INCLUDE_DIR-NOTFOUND

alphapibeta, 2020-08-16:
> Thanks @ceccocats ,
> I have solved this problem according to your suggestion.

Hello @luckcodingdog, can you please share the steps, as I am also facing the same issue and, I too have setup my Tensorrt via 4.3 part of the guide of https://docs.nvidia.com/deeplearning/tensorrt/install-guide/ 
```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda-10.0 (found suitable version "10.0", minimum required is "9.0") 
-- Found CUDNN: /usr/local/cuda-10.0/lib64/libcudnn.so
-- Found CUDNN include: /usr/local/cuda-10.0/include
-- Found NVINFER: /home/turing/Downloads/TensorRT-7.0.0.11/lib
-- Found NVINFER include: NVINFER_INCLUDE_DIR-NOTFOUND
-- Found CUDNN: /usr/local/cuda-10.0/lib64/libcudnn.so  
-- Found OpenCV: /usr/local (found version "4.2.0") 
install dir:/usr/local
-- Configuring done
WARNING: Target "test_yolo3_512" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_simple" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_yolo2tiny" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_rtinference" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_yolo3_berkeley" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_yolo3" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_yolo2_voc" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_csresnext50-panet-spp" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_yolo2" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_csresnext50-panet-spp_berkeley" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "tkDNN" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_yolo3tiny" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_yolo3_coco4" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_imuodom" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_dla34" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_mnist" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "map_demo" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_mnistRT" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_mobilenetv2ssd" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_yolo3_flir" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_yolo3tiny_512" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
[... cut ...]
WARNING: Target "test_resnet101_cnet" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "demo" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
WARNING: Target "test_viz_yolo3" requests linking to directory "/home/turing/Downloads/TensorRT-7.0.0.11/lib".  Targets may link only to libraries.  CMake is dropping the item.
-- Generating done
-- Build files have been written to: /home/turing/expriments/tkDNN/build

```
Thanks in advance.

luckcodingdog, 2020-08-17:
You can copy the include and lib files of tensorrt to the include/lib path of cuda,
then you can get the  lib of " NVINFER include"

mive93, 2020-09-11:
Closing for now,
feel free to reopen.

closed 2020-09-11 by mive93

masip85, 2021-10-06:
can  this https://github.com/ceccocats/tkDNN/issues/62#issuecomment-658019576 be achieved trought cmake flags?