ceccocats/tkDNN #64: Funny (YOLOv3) bounding-box on Windows

jstumpin, 2020-07-01:
After converting the layers into *.rt via _test_yolov3.exe_ with "TKDNN_MODE = FP16" and _yolov3.cfg_ with "width/height=320", here's the output from _demo.exe_:
[image]

Another observation is, only a particular input dim is supported (can be executed error-free either via _test_yolov3.exe_ or _demo.exe_), where in this case "320x320" (the provided _yolov3.cfg_ was originally [416x416](https://github.com/ceccocats/tkDNN/blob/04602f395236d993d79cff7d1be5ced3ae738b0b/tests/darknet/cfg/yolo3.cfg#L8) and it will end up as "Cuda failure: an illegal memory access was encountered" at [NetworkRT.cpp:207](https://github.com/ceccocats/tkDNN/blob/04602f395236d993d79cff7d1be5ced3ae738b0b/src/NetworkRT.cpp#L207), if I commented that line, the error would fallback to [cudaMemcpyAsync](https://github.com/ceccocats/tkDNN/blob/04602f395236d993d79cff7d1be5ced3ae738b0b/src/NetworkRT.cpp#L206), which could indicate something wrong with the output layer). Ironically, the provided weights whose input is "416x416" failed to run whereas "320x320" is able to, albeit the scaling mishap. 

Any pointers?

Thanks.

ceccocats, 2020-07-01:
[image]
this is the result with yolo3 320x320 using the weights of yolo3 416x416 (still, using the weights of a different dimension is not recommended, it is better to retrain).
tkDNN has a lot of issue on Windows but sorry I'm not going to test all on Windows soon.

ceccocats, 2020-07-01:
[image]
Also fp16 is ok

zpmmehrdad, 2020-07-01:
@jstumpin Hi,

I used TensorRT v 7.0.0.11 on windows, but I got an error after the deserialization. Did you have a problem with this version?

jstumpin, 2020-07-03:
@zpmmehrdad,

I think it is resolvable by performing this [step](https://github.com/ceccocats/tkDNN/issues/28#issuecomment-643768379). I'm using TensorRT v6.01 but still had that deserialization error initially.

jstumpin, 2020-07-10:
Upgraded to TensorRT v7.1.3.4, CUDA v11.0.2, CUDNN v8.0.1.13, a little hiccup during building, nevertheless same outcome.

arnaud-nt2i, 2020-09-11:
@jstumpin @JoeCool90 @zpmmehrdad 

Hi, there! 
Did one of you manage to make YoloV4 work on tkDNN framework on windows 10 ?
For deployment reasons, Linux is not possible for us...
Any int would be appreciated.

Otherwise, we might test OpenCV DNN

JoeCool90, 2020-09-11:
See [https://github.com/ceccocats/tkDNN/issues/54]

Basically I rolled my own solution borrowing from a couple of different frameworks like tkDNN, tensorRTX.  tensorRTX is the most up to date in the way it uses the nvidia tensorRT framework but has its own issues (hard coded and missing scaling).  Additionally, the yolov4 source makes changes that subtly deviate from the implementations above so rolling your own version is basically the only way to stay remotely up to date.

Unfortunately the tensorRT docs are missing just enough information in them, and their example code is often outdated just enough, that implementing your own versions are non-trivial.  I recommend tensorRTX for understanding how to use tensorRT, tkDNN for passing and ticking all the yolo feature boxes, and yolov4 source for updates to how some layers are implemented.

arnaud-nt2i, 2020-09-11:
ok, Thank you very much Joe !
I will try this https://github.com/enazoe/yolo-tensorrt
And compare it with OpenCV DNN.

dsgh2, 2020-09-26:
How did you build on Windows? I have many errors while building project generated by cmake

perseusdg, 2021-04-24:
tkDNN is now supported on windows on the current master branch at an experimental level

closed 2021-11-25 by mive93