ceccocats/tkDNN #67: Feature-request: TensorRT 7.1.3 supporting

mochechan, 2020-07-07:
Current tkDNN supports TensorRT 6.0.1. However, newer TensorRT was released.
Is it possible that tkDNN supports TensorRT 7.1.3?

mive93, 2020-07-07:
Hi @mochechan ,

Have you tried it? Because we actually support 7.1.0 and I don't think it should be that much different.

mochechan, 2020-07-08:
In my evaluation, to install the following TensorRT 6.0.1 libraries can compile the tkDNN without error. The compiled result can be executed as well.
**apt-get install -y  libnvinfer6=6.0.1-1+cuda10.2 libnvinfer-dev=6.0.1-1+cuda10.2 libnvinfer-plugin6=6.0.1-1+cuda10.2 libeigen3-dev libopencv-dev**

However, to install the following TensorRT 7.1.3 libraries occurs errors when compiling the tkDNN.
**apt-get install -y  libnvinfer7=7.1.3-1+cuda10.2 libnvinfer-dev=7.1.3-1+cuda10.2 libnvinfer-plugin7=7.1.3-1+cuda10.2 libeigen3-dev libopencv-dev**

The error messages are described as the follows.
In file included from /root/tkDNN/include/tkDNN/Layer.h:6:0,
                 from /root/tkDNN/src/Conv2d.cpp:3:
/root/tkDNN/src/Conv2d.cpp: In member function 'void tk::dnn::Conv2d::initCUDNN(bool)':
/root/tkDNN/src/Conv2d.cpp:68:57: error: 'CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST' was not declared in this scope
                                                         CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST, 0, &bwAlgo) );
                                                         ^
/root/tkDNN/src/Conv2d.cpp:66:9: note: in expansion of macro 'checkCUDNN'
         checkCUDNN( cudnnGetConvolutionBackwardDataAlgorithm(net->cudnnHandle,
         ^~~~~~~~~~
/root/tkDNN/src/Conv2d.cpp:68:57: note: suggested alternative: 'CUDNN_CONVOLUTION_BWD_DATA_ALGO_FFT'
                                                         CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST, 0, &bwAlgo) );
                                                         ^
/root/tkDNN/src/Conv2d.cpp:66:9: note: in expansion of macro 'checkCUDNN'
         checkCUDNN( cudnnGetConvolutionBackwardDataAlgorithm(net->cudnnHandle,
         ^~~~~~~~~~
/root/tkDNN/src/Conv2d.cpp:66:21: error: 'cudnnGetConvolutionBackwardDataAlgorithm' was not declared in this scope
         checkCUDNN( cudnnGetConvolutionBackwardDataAlgorithm(net->cudnnHandle,
                     ^
/root/tkDNN/include/tkDNN/utils.h:68:9: note: in definition of macro 'checkCUDNN'
     if (status != CUDNN_STATUS_SUCCESS) {                              \
         ^~~~~~
/root/tkDNN/src/Conv2d.cpp:66:21: note: suggested alternative: 'cudnnGetConvolutionBackwardDataAlgorithm_v7'
         checkCUDNN( cudnnGetConvolutionBackwardDataAlgorithm(net->cudnnHandle,
                     ^
/root/tkDNN/include/tkDNN/utils.h:68:9: note: in definition of macro 'checkCUDNN'
     if (status != CUDNN_STATUS_SUCCESS) {                              \
         ^~~~~~
/root/tkDNN/src/Conv2d.cpp:79:57: error: 'CUDNN_CONVOLUTION_FWD_PREFER_FASTEST' was not declared in this scope
                                                         CUDNN_CONVOLUTION_FWD_PREFER_FASTEST, 0, &algo) );
                                                         ^
/root/tkDNN/include/tkDNN/utils.h:68:9: note: in definition of macro 'checkCUDNN'
     if (status != CUDNN_STATUS_SUCCESS) {                              \
         ^~~~~~
/root/tkDNN/src/Conv2d.cpp:79:57: note: suggested alternative: 'CUDNN_CONVOLUTION_BWD_FILTER_ALGO_3'
                                                         CUDNN_CONVOLUTION_FWD_PREFER_FASTEST, 0, &algo) );
                                                         ^
/root/tkDNN/include/tkDNN/utils.h:68:9: note: in definition of macro 'checkCUDNN'
     if (status != CUDNN_STATUS_SUCCESS) {                              \
         ^~~~~~
/root/tkDNN/src/Conv2d.cpp:77:21: error: 'cudnnGetConvolutionForwardAlgorithm' was not declared in this scope
         checkCUDNN( cudnnGetConvolutionForwardAlgorithm(net->cudnnHandle,
                     ^
/root/tkDNN/include/tkDNN/utils.h:68:9: note: in definition of macro 'checkCUDNN'
     if (status != CUDNN_STATUS_SUCCESS) {                              \
         ^~~~~~
/root/tkDNN/src/Conv2d.cpp:77:21: note: suggested alternative: 'cudnnGetConvolutionForwardAlgorithm_v7'
         checkCUDNN( cudnnGetConvolutionForwardAlgorithm(net->cudnnHandle,
                     ^
/root/tkDNN/include/tkDNN/utils.h:68:9: note: in definition of macro 'checkCUDNN'
     if (status != CUDNN_STATUS_SUCCESS) {                              \
         ^~~~~~
CMakeFiles/tkDNN.dir/build.make:120: recipe for target 'CMakeFiles/tkDNN.dir/src/Conv2d.cpp.o' failed
make[2]: *** [CMakeFiles/tkDNN.dir/src/Conv2d.cpp.o] Error 1
CMakeFiles/Makefile2:436: recipe for target 'CMakeFiles/tkDNN.dir/all' failed
make[1]: *** [CMakeFiles/tkDNN.dir/all] Error 2
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2

I haven't test TensorRT 7.1.0 yet.

minhtcai, 2020-07-13:
Hi, do you have any updates on that? I got the same problem.

ceccocats, 2020-07-13:
This is a cudnn error, not tensorrt.
Can you check what version of cuDNN?
we support until 8.0.0, I bet you have 8.0.1

minhtcai, 2020-07-14:
Yeah that's the 8.0.1. I then rolled back to TensorRT 6.0.1 and everything works perfectly!

alexanderfrey, 2020-07-15:
For Xavier NX you can not roll back since everything is part of JetPack 4.4. Any updates on this ?

closed 2020-09-11 by mive93