ceccocats/tkDNN #69: Example-Request: tkDNN YOLOv4 object detection on custom data

Seioch, 2020-07-07:
I would love a guide on training your own YOLOv4 detector. Something like taking the weights from a darknet trained detector and connecting it to tkDNN for faster inference.

ceccocats, 2020-07-13:
Have you tried the exporter?
https://github.com/ceccocats/tkDNN#1export-weights-from-darknet

You just have to run the exporter in your weights and copy the cfg and weights in tkDNN

Seioch, 2020-07-13:
Yup, though the README wasn't explicit that I had to copy the cfg, weights, and debug files into the yolo4 folder, as well as recompile yolo4.cpp to have the right paths to the cfg and names. Did figure it out in the end though. :)

closed 2020-07-13 by Seioch

alphapibeta, 2020-07-22:
> Yup, though the README wasn't explicit that I had to copy the cfg, weights, and debug files into the yolo4 folder, as well as recompile yolo4.cpp to have the right paths to the cfg and names. Did figure it out in the end though. :)

hope [this](https://github.com/ceccocats/tkDNN/issues/52#issuecomment-662473806) helps anyone whos is looking for to run their custom object detector.