ceccocats/tkDNN #71: Segmentation fault in test_yolo4tiny

jerrywgz, 2020-07-09:
Thanks for your great work! Now I want to test yolo4tiny on my own environment. I followed the instruction from document and build successfully. When I tried to run  ```./test_yolo4tiny```, I received segmentation fault as follows:
[image]

My environment is V100 CUDA10.1 CUDNN7.6.5 TensorRT6.01. How can I solve this problem?

mive93, 2020-07-10:
Hi @jerrywgz,
which GPU or board are you using?

jerrywgz, 2020-07-10:
> Hi @jerrywgz,
> which GPU or board are you using?

NVIDIA Tesla V100 GPU

jerrywgz, 2020-07-15:
anyone met such problem? I guess it is caused in createBuilderConfig from TensorRT. Is there something wrong when installing TensorRT?

mive93, 2020-07-15:
Can you turn on debug please, and copy here that output. Thanks.

```
cmake .. -DDEBUG=True
make
```

gabenell, 2020-07-24:
I had a segfault at the same place after doing `apt upgrade` recently, which updated some CUDA libs and headers on my Jetson NX. After deleting the previously-generated .rt files from my build directory, the tests worked (as well as my code which uses tkDNN)

closed 2020-09-11 by mive93