ceccocats/tkDNN #73: Error reading file while converted darknet exported weights of yolov4-tiny

naufil-hassan-confiz, 2020-07-10:
While converting darknet exported weights to generate *.rt file; by running ./test_yolov4tiny I got this error 
_Error reading file yolo4tiny/layers/c29.bin with n of float: 130560 seek: 0 size: 522240_

It should be noted that I trained this darknet network on my own data and made it a 1-class detector; that's all I have changed.
If I donot provide it the exported weights, the executable downloads the required config/weights and then it generates the *.rt file successfully.

I am using following setup for this experiment
>  Cuda 10.0
> Cudnn 7.6
> TensorRT 6.0.1.5
> TitanXP

Seioch, 2020-07-11:
I ran into this problem. The way I fixed it was ensuring that the cfg files during the export was the same as the one I used for training.

ceccocats, 2020-07-13:
Like @Seioch  said you must change the cfg in:
https://github.com/ceccocats/tkDNN/blob/master/tests/darknet/cfg/
Or open your cfg in the test file

rod-hendricks, 2020-07-16:
Not sure if anyone still having issues with this but ran into this also. You need to keep the cfg file in sync with all 3 sources that needs it (training, exporting and creation of rt file). I was only keeping the 1st 2 in sync but did not immediately realize that the 3rd was always pointing to the default cfg and names file in the test/darknet folder.

closed 2020-09-11 by mive93