ceccocats/tkDNN #74: cudnn 8.0 support

samsonadmin, 2020-07-13:
error: ‘CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST’ was not declared in this scope
                                                         CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST, 0, &bwAlgo) );

error: ‘CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST’ was not declared in this scope
                                                         CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST, 0, &bwAlgo) );

AastaLLL gave some help:
https://raw.githubusercontent.com/AastaNV/JEP/master/script/topics/0001-fix-for-cudnn_v8-limited-memory-to-default-darknet-s.patch

ceccocats, 2020-07-13:
Actually cudnn8.0.0 support was added with this commit:
https://github.com/ceccocats/tkDNN/commit/20303ac32ef8e28ca0ff94be633478d7755ae307
You are probably using 8.0.1
Nvidia changes a lot, it is not simple to keep updated.
We decided to keep updated with the versions shipped with Jetpack. The current release is:
Jetpack 4.4 (CUDA 10.2, CUDNN 8.0.0, tensorrt 7.1.0

ahmedius2, 2020-07-16:
I am trying to make it work on Jetson NX which has CUDNN 8.0.0 but it does not compile without modification. I modified the code and was able to compile, [here is the modifications I have made](https://drive.google.com/file/d/1Lj7CiwX7rWRV5kkFksE1dvfxYuXZ6ivr/view?usp=sharing).

However, all demo programs I have tried fails with the same error (CUDNN_STATUS_NOT_SUPPORTED). Maybe I couldnt modify it correctly. Here is the output of ./test_yolo3:
ubuntu@xavier-nx:~/ahmet/tkDNN/build$ ./test_yolo3
Not supported field: batch=32
Not supported field: subdivisions=32
Not supported field: momentum=0.9
Not supported field: decay=0.0005
Not supported field: angle=0
Not supported field: saturation = 1.5
Not supported field: exposure = 1.5
Not supported field: hue=.1
Not supported field: learning_rate=0.001
Not supported field: burn_in=1000
Not supported field: max_batches = 500200
Not supported field: policy=steps
Not supported field: steps=400000,450000
Not supported field: scales=.1,.1
New NETWORK (tkDNN v0.5, CUDNN v8)
Reading weights: I=3 O=32 KERNEL=3x3x1
Reading weights: I=32 O=64 KERNEL=3x3x1
Reading weights: I=64 O=32 KERNEL=1x1x1
Reading weights: I=32 O=64 KERNEL=3x3x1
Reading weights: I=64 O=128 KERNEL=3x3x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=64 O=128 KERNEL=3x3x1
Reading weights: I=128 O=64 KERNEL=1x1x1
Reading weights: I=64 O=128 KERNEL=3x3x1
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=512 KERNEL=3x3x1
Reading weights: I=512 O=256 KERNEL=1x1x1
CUDNN failure: CUDNN_STATUS_NOT_SUPPORTED
/home/ubuntu/ahmet/tkDNN/src/Conv2d.cpp:81
Aborting...

ahmedius2, 2020-07-16:
I had the same problem as you and modified the code to be able to compile and run with cuDNN 8.0.0 on jetson nx. Here is the patch: https://drive.google.com/file/d/17f_i7uYClNxyCrL2_QLQXNMDp5OhShKE/view?usp=sharing

commit 6a68f19b (2020-07-16): Fix patch from @ahmedius2 , tkDNN now supports CUDNN 8.0.1 (Fix #74)

	       Francesco Gatti <gattifrancesco@hotmail.it>
closed 2020-07-16 by mive93