ceccocats/tkDNN #75: About inference with BATCHSIZE > 1

ferociousricky, 2020-07-14:
@ceccocats :
Hi I'm Rex. I am having problem with creating the videos "result.avi", when i set the batch size larger than 1. Can you help me out or correct my procedure, if i am wrong.
My procedure:
=================================================================================================
1. git clone https://github.com/ceccocats/tkDNN
2. cmake  ../
3. make 
4. git clone https://git.hipert.unimore.it/fgatti/darknet.git # Export weights from darknet.
5. make 
6. mkdir layers debug
7. ./darknet export <path-to-cfg-file> <path-to-weights> layers
8.export TKDNN_MODE=FP32
9.export TKDNN_BATCHSIZE=16
10. ./test_yolo4
11. /test_rtinference yolo4_fp32.rt 4
12. ./demo yolo4_fp32.rt ../demo/yolo_test.mp4 y 80 32 0
================================================================================================

Thanks,
BR Rex

mive93, 2020-07-14:
Hi @ferociousricky,
Yes it is correct. In the demo there is an if that let you save the video only for bacth = 1 (line 123 https://github.com/ceccocats/tkDNN/blob/master/demo/demo/demo.cpp).
The code should be changed (not only the if) if you want it to work correctly. However, the output video would be the same wrt the one created with batch=1.

ferociousricky, 2020-07-15:
@mive93 
Thanks for help and this fantastic project!! :")

closed 2020-07-17 by ferociousricky