ceccocats/tkDNN #76: tensorrt in multi-thread

wu-ruijie, 2020-07-14:
I run tkDNN in 2 threads, I Create 2 DetectionNN( as Yolo3Detection),  and so it's 2 different contextRT, 2 different stream,
in NetworkRT::infer(), I test spend time with contextRT->enqueue(), in 1 thread, it spend 3ms, in 2 thread ,it spend more than 10ms, although I Comments checkCuda(cudaStreamSynchronize(stream)) in NetworkRT::infer()， but there is nothing change.

wu-ruijie, 2020-07-14:
when I run tkDNN in 2 process/instance，every instance spend 3ms with contextRT->enqueue() in NetworkRT::infer()

mive93, 2020-07-15:
Hi @wu-ruijie 

sorry, I don't really get your problem. Can you explain better?

wu-ruijie, 2020-07-16:
hi @mive93 

my code like this:

void *func(void *ptr)
{
   int id = *(int *)ptr;
   tk::dnn::Yolo3Detection yolo;
   tk::dnn::CenternetDetection cnet;
   tk::dnn::MobilenetDetection mbnet;
   tk::dnn::DetectionNN *detNN = &yolo;

   detNN->init(net, n_classes, n_batch);
   auto startTime = std::chrono::high_resolution_clock::now();
   detNN->update(batch_dnn_input, n_batch);
   auto endTime = std::chrono::high_resolution_clock::now();
   float totalTime = std::chrono::duration<float, std::milli>(endTime - startTime).count();
   printf("thread %d time=%f ms\n", id, totalTime);
   detNN->draw(batch_dnn_input/*batch_frame*/);
}

int main()
{
   int id[2]={1,2};
   pthread_t a, b;
   pthread_create(&a, NULL, func, &id[0]);
   pthread_create(&b, NULL, func, &id[1]);
}

the detect result is correct, if only one thread the totalTime about 10ms, but 2 thread,  thread 1 time ~ 17 ms, thread 2 time ~18 ms, while i thank every thread time ~10 ms

I find the problem is here:  NetworkRT.cpp/NetworkRT::infer/contextRT->enqueue, but I have no idea to fix this bug
thanks very much!

mive93, 2020-07-17:
I see.
It is completely normal. It has to process 2 images instead of 1.
I have made some tests myself, here are the results using yolo4 and the video yolo_test.mp4 that you can find in this repo

| configuration  | time for update for thread (ms) | 
|:-:|:-:|
| 1 thread, batch = 1 | ~9 |
| 2 thread, batch = 1 | ~17 |
| 1 thread, batch = 2 | ~17| 

So as you can see, having 2 threads lead to the same performances of having batch = 2. There is more work to do, that's all.

kunyao2015, 2020-08-11:
Is there any solution?

mive93, 2020-09-11:
What do you mean any solution? 
It's how the GPU works.

closed 2021-01-04 by mive93