ceccocats/tkDNN #77: How to use this project as .so file?

jzchenriver, 2020-07-14:
I want to use tkDNN to accelerate the yolov4-tiny in QT.
After compiling, I got libkernels.so and libtkDNN.so.
When using them, I need to perform the declaration.
In the original yolov4, there is a yolo_v2_class.hpp, which includes all the .h files in the project. 
In the tkDNN, Is there a similar file?
Thank you.

ceccocats, 2020-07-14:
Hi,
you can install tkDNN in your sistem and use it as a normal library with:
```
sudo make install
```
Or you can use tkDNN as a cmake submodule.

once is installed is sufficent to include like this:
```
#include "tkDNN/tkdnn.h"                // tkdnn core
#include "tkDNN/DarknetParser.h" // darknet parser utils
```

ghimiredhikura, 2020-07-14:
hi @jzchenriver, 

Ofcourse @ceccocats approach is the primary one. But if you want to link tkdnn *.so files explicitely in your CMake here is template for CMakeLists.txt. I did sucessfully use *.so file in similar way. 

**CMakeLists.txt**
```ruby
cmake_minimum_required(VERSION 3.5)

project(tkdnn_demo)
set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} ${CMAKE_CURRENT_SOURCE_DIR}/cmake)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -fPIC -Wno-deprecated-declarations -Wno-misleading-indentation -Wno-sign-compare -Wno-unused-variable")
add_definitions("-Wall")
add_definitions(-DUNICODE -D_UNICODE)

# CUDA
find_package(CUDA 9.0 REQUIRED)
SET(CUDA_SEPARABLE_COMPILATION ON)
set(CUDA_NVCC_FLAGS ${CUDA_NVCC_FLAGS} --maxrregcount=32)
include_directories(${CUDA_INCLUDE_DIRS})

# CUDNN
find_package(CUDNN REQUIRED)
include_directories(${CUDNN_INCLUDE_DIR})

include_directories(${CMAKE_CURRENT_SOURCE_DIR}/include)

# Used for I/O and other image processing operations
find_package(OpenCV REQUIRED)

# TKDNN - object detection 
include_directories(${CMAKE_CURRENT_SOURCE_DIR}/tkdnn/include)
file(GLOB TKDNN_LIBS ${CMAKE_CURRENT_SOURCE_DIR}/tkdnn/lib/*.so)

set(LIBS cudart ${OpenCV_LIBS} ${TKDNN_LIBS})

# set source files 
file(GLOB SRC ${CMAKE_CURRENT_SOURCE_DIR}/src/*.cpp)

# executable
add_executable(tkdnn_demo ${SRC})
target_link_libraries(tkdnn_demo ${LIBS})
```
Note:  
tkdnn include files: https://github.com/ceccocats/tkDNN/tree/master/include/  
tkdnn .so file: look into build folder.   

may be @ceccocats wants to beautify above CMakeLists.txt with only required lines. ^^

jzchenriver, 2020-07-14:
I met this problem:
1    error: widget.o: undefined reference to symbol '_ZN2cv11getTextSizeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEidiPi'
2   /usr/lib/aarch64-linux-gnu/libopencv_imgproc.so.4.1:-1: error: error adding symbols: DSO missing from command line
3  : error: collect2: error: ld returned 1 exit status

ghimiredhikura, 2020-07-14:
Install opencv first with this script:   
https://github.com/ceccocats/tkDNN/blob/master/scripts/install_OpenCV4.sh

closed 2020-07-20 by jzchenriver

mochechan, 2022-05-06:
I met a similar problem if using jetpack 4.6.1(jetson xavier nx). But everything is fine if using jetpack 4.5.1.

undefined reference to symbol '_ZN8nvinfer121RegionRTPluginCreatorC1Ev'
//usr/local/lib/libkernels.so: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status

What package should I install?