ceccocats/tkDNN #83: Jetston TX2/JP4.3 : Segfault when running ./test_yolo4

MI-LA01, 2020-07-16:
Hello, 
Just built tkdnn on a tx2 ; and tried ./test_yolo4 however : 
```
262 ActivationLeaky   13 x   13, 1024  ->   13 x   13, 1024
263 Conv2d            13 x   13, 1024  ->   13 x   13,  512
264 ActivationLeaky   13 x   13,  512  ->   13 x   13,  512
265 Conv2d            13 x   13,  512  ->   13 x   13, 1024
266 ActivationLeaky   13 x   13, 1024  ->   13 x   13, 1024
267 Conv2d            13 x   13, 1024  ->   13 x   13,  255
268 Yolo              13 x   13,  255  ->   13 x   13,  255
===========================================================

GPU free memory: 3897.8 mb.
New NetworkRT (TensorRT v6.01)
Float16 support: 1
Int8 support: 0
DLAs: 0
create execution context
Segmentation fault (core dumped)
```
- I ran through ssh with the -X flag .. just in case X is used ;
- Cuda 10.0
- Cudnn 7.6
- TensorRT 6.01

 any advices are welcome

mive93, 2020-07-17:
Hi @MI-LA01 
Can you turn on debug please, and copy here that output. Thanks.

```
cmake .. -DDEBUG=True
make
```

MI-LA01, 2020-07-17:
Hello @mive93 : here it is : 
```
PU free memory: 4372.63 mb.
New NetworkRT (TensorRT v6.01)
Float16 support: 1
Int8 support: 0
DLAs: 0
TENSORRT LOG: INVALID_ARGUMENT: Cannot deserialize with an empty memory buffer.
TENSORRT LOG: INVALID_CONFIG: Deserialize the cuda engine failed.
create execution context
Segmentation fault (core dumped)
```

mive93, 2020-07-17:
Hi @MI-LA01 
Is it possible that you already have a .rt file (given that it says ```Deserialize the cuda engine failed```)? 
If so, remove it and try the test again.

MI-LA01, 2020-07-21:
hi @mive93 - You were right, removing the previous generated engine, compiling w/ debug and relaunching the yolov4 bin make it works.

closed 2020-07-21 by MI-LA01