ceccocats/tkDNN #84: yolo4.cpp:3:10: fatal error: tkdnn.h: No such file or directory

Divyam10, 2020-07-17:
I am trying to compile yolo4.cpp using 
`sudo g++ yolo4.cpp
`
but i am getting the following error 
```
yolo4.cpp:3:10: fatal error: tkdnn.h: No such file or directory
 #include "tkdnn.h"
          
compilation terminated.
```
Ubuntu 18
Cuda 10.0 
tensorRT 6.0.1.8

mive93, 2020-07-17:
Hi @Divyam10 
That's not how you compile the project.
Follow the readme: https://github.com/ceccocats/tkDNN#how-to-compile-this-repo

Divyam10, 2020-07-17:
Hi @mive93 I have a confusion if you can help me with that.
I compiled the project as described in the repo, mentioning below as well.
```
git clone https://github.com/ceccocats/tkDNN
cd tkDNN
mkdir build
cd build
cmake .. 
make
```
Then I exported the weights using the below

```
git clone https://git.hipert.unimore.it/fgatti/darknet.git
cd darknet
make
mkdir layers debug
./darknet export <path-to-cfg-file> <path-to-weights> layers
```
Then I put debug and layers in one folder and copied it to tkdnn/tests/darkenet

I changed the path in yolo4.cpp

now how to do i generate the .rt file for it?
I don't quite understand this step mentioned in the repo

> Create a new test and define the network, layer by layer using the weights extracted and the output to check the results.

Am I suppose to run the yolo4.cpp file in tests/darknet, if yes then how?
because compiling the same using gcc gave me the above error.

I have also tried to put the folder with layers and debug in tkdnn/build and recompiling it but no luck. 
Maybe I am doing something utterly wrong.

mive93, 2020-07-17:
You are using this project in the wrong way. 
Do not use g++ o gcc to compile single files. That's not how you compile a c++ project. 

Darknet uses a Makefile for the compilation, we use cmake instead. 
To correctly compile tkDNN you should run (from the build folder)
```
cmake ..
make
```
You have to be sure that cmake does not give you errors. And that make completes with 100%.
Moreover, be sure you satisfy all the dependencies mentioned in the README.

When you have successfully compiled, just run
```
./test_yolo4
```
It will automatically download the weights and will create the .rt for yolov4 size 416x416. 
The export phase is needed only if you want different networks from the ones you can already find in this repo. 

Once you have the rt file (should be called yolo4_fp32.rt) you can run the demo just with
```
./demo yolo4_fp32.rt
```

All the other details are in the README.

Divyam10, 2020-07-17:
Hi @mive93. 
Thank you it worked, I clearly did not understand the process. 

Also wanted to address one more issue I came across with the download site

> https://cloud.hipert.unimore.it/s/d97CFzYqCPCp5Hg/download

I could not access this site from my region, so while executing it showed an error of no zip found.
I had to solve this by manually  downloading the files using Tor/Vpn and copy and pasting to the folder. ( Also realised later I could have just copied layers and debug folders directly which I had exported above xD. )

Ps - Thanks for your help @mive93

closed 2020-07-17 by Divyam10