ceccocats/tkDNN #86: Cuda Failure: Invalid Arguement

Teresito, 2020-07-19:
Hi,

I am trying to run Mobilenetv2-SSD using your repo on a Jetson Nano.

These were the output from the terminal.

My command
`./demo mobilenetv2ssd_fp32.rt ../demo/yolo_test.mp4 m`

Output from terminal
`detection
mobilenetv2ssd_fp32.rt
New NetworkRT (TensorRT v7.13)
Float16 support: 1
Int8 support: 0
DLAs: 0
create execution context
Input/outputs numbers: 5
input idex = 0 -> output index = 4
Data dim: 1 3 300 300 1
Data dim: 1 3000 1 4 1
RtBuffer 0   dim: Data dim: 1 3 300 300 1
RtBuffer 1   dim: Data dim: 1 126 1 1 1
RtBuffer 2   dim: Data dim: 1 24 1 1 1
RtBuffer 3   dim: Data dim: 1 21 3000 1 1
RtBuffer 4   dim: Data dim: 1 3000 1 4 1
camera started
Cuda failure: invalid argument
/home/computervision/tkDNN/src/MobilenetDetection.cpp:255
Aborting...
`
Please let me know if you are requiring any other information to further provide a solution to this problem. 

Thank you

mive93, 2020-07-21:
Hi, 
that model has been trained on VOC, so you should also change the clesses number:

```
./demo mobilenetv2ssd_fp32.rt ../demo/yolo_test.mp4 m 20

```

Teresito, 2020-07-22:
Thank you for the reply, it ran smoothly :)

closed 2020-07-22 by Teresito