ceccocats/tkDNN #87: yolov4-custom.cfg ,when run test_yolo4 , error.

bjthemost, 2020-07-20:
Reading weights: I=128 O=256 KERNEL=3x3x1
Reading weights: I=256 O=128 KERNEL=1x1x1
Reading weights: I=128 O=256 KERNEL=3x3x1
Error reading file yolo4/layers/c137.bin with n of float: 294912 seek: 0 size: 1179648
-----------
my custom file is:

[net]
# Testing
#batch=1
#subdivisions=1
# Training
batch=1 #16
subdivisions=1 #8
width=608
height=608
channels=3
momentum=0.949
decay=0.0005
angle=0
saturation = 1.5
exposure = 1.5
hue=.1
flip=0

learning_rate=0.001
burn_in=1000
max_batches = 90000
policy=steps
steps=72000,81000
scales=.1,.1

#cutmix=1
mosaic=1

#:104x104 54:52x52 85:26x26 104:13x13 for 416

[convolutional]
batch_normalize=1
filters=32
size=3
stride=1
pad=1
activation=mish

# Downsample

[convolutional]
batch_normalize=1
filters=64
size=3
stride=2
pad=1
activation=mish

[convolutional]
batch_normalize=1
filters=64
size=1
stride=1
pad=1
activation=mish

[route]
layers = -2

[convolutional]
batch_normalize=1
filters=64
size=1
[... 1060 lines cut ...]

[convolutional]
size=1
stride=1
pad=1
filters=150
activation=linear

[yolo]
mask = 6,7,8
anchors = 12, 16, 19, 36, 40, 28, 36, 75, 76, 55, 72, 146, 142, 110, 192, 243, 459, 401
classes=45
num=9
jitter=.3
ignore_thresh = .7
truth_thresh = 1
random=1
scale_x_y = 1.05
iou_thresh=0.213
cls_normalizer=1.0
iou_normalizer=0.07
iou_loss=ciou
nms_kind=greedynms
beta_nms=0.6
max_delta=5

mive93, 2020-07-20:
Hi @bjthemost ,
how did you export the weights?

bjthemost, 2020-07-20:
> Hi @bjthemost ,
> how did you export the weights?

yes, darknet export cfg/yolov4.cfg yolov4.weights ./layers ,    like this

bjthemost, 2020-07-20:
I m fix it .....
 std::string bin_path  = "yolo4";
    std::vector<std::string> input_bins = {
        bin_path + "/layers/input.bin"
    };
    std::vector<std::string> output_bins = {
        bin_path + "/debug/layer139_out.bin",
        bin_path + "/debug/layer150_out.bin",
        bin_path + "/debug/layer161_out.bin"
    };
    std::string wgs_path  = bin_path + "/layers";
    std::string cfg_path  = std::string(TKDNN_PATH) + "/tests/darknet/cfg/yolo4.cfg";
    std::string name_path = std::string(TKDNN_PATH) + "/tests/darknet/names/coco.names";
-----
this is HARD CODE
-------------------------------------------
but  other question

Not supported field: anchors = 12, 16, 19, 36, 40, 28, 36, 75, 76, 55, 72, 146, 142, 110, 192, 243, 459, 401
Not supported field: jitter=.3
Not supported field: ignore_thresh = .7
Not supported field: truth_thresh = 1
Not supported field: iou_thresh=0.213
Not supported field: cls_normalizer=1.0
Not supported field: iou_normalizer=0.07
Not supported field: iou_loss=ciou
Not supported field: nms_kind=greedynms
Not supported field: beta_nms=0.6
Not supported field: max_delta=5
-------

zhengjiawen, 2020-07-21:
I have same problem, could you tell me how to solve it?

mive93, 2020-07-21:
Hi @bjthemost,

yes, this part is hard coded
```
std::vectorstd::string output_bins = {
bin_path + "/debug/layer139_out.bin",
bin_path + "/debug/layer150_out.bin",
bin_path + "/debug/layer161_out.bin"
};
```
However, is just for the test, to check layer by layer. 
It is normal that the parser print the not supported fields. It's all ok, there should be no problems. 
Have you tried using the rt file with the demo and check the performance?

andressa-suiti, 2020-09-11:
Hi @bjthemost ,
how did you solve this problem? I used ./darknet export <path-to-cfg-file> <path-to-weights> layers to export the layers and had the same issue

mive93, 2020-09-15:
@bjthemost you have to be sure to copy the cfg and change the output_bins. 
Let me know if you solved it.

bjthemost, 2020-09-17:
sorry ,@mive93, 
 
The solution is to convert bin files on the target board (Jetson Xavier NX) ,And modify the hard coding in the source code

sanketgoyal, 2020-11-17:
> The solution is to convert bin files on the target board (Jetson Xavier NX) ,And modify the hard coding in the source code

Hello @bjthemost, what should be the new modification? How do I find the correct output layer numbers from my custom model?

mive93, 2021-01-04:
Check this: https://github.com/ceccocats/tkDNN/issues/152#issuecomment-738680078

Closing for now, feel free to reopen.

closed 2021-01-04 by mive93