ceccocats/tkDNN #89: Error when cmake - Missing CUDA_nvinfer_LIBRARY

minhtcai, 2020-07-21:
Hi, I've trying to cmake but I got this error:

################

-- Found CUDNN: /usr/local/cuda-10.0/lib64/libcudnn.so
-- Found CUDNN include: /usr/local/cuda-10.0/include
-- Found OpenCV: /usr (found version "3.2.0")
-- Found OpenCV: /home/minh/opencv/build (found version "4.4.0")
install dir:/usr/local
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_nvinfer_LIBRARY
linked by target "tkDNN" in directory /home/minh/tkDNN

-- Generating done
CMake Generate step failed. Build files cannot be regenerated correctly.

#################
My system:
Ubuntu 18.04
GTX970M
CUDA 10.0
CUDNN 7.6.5
OpenCV 4.4
TensorRT 7.0.0-1+cuda10.0 (install using Debian https://docs.nvidia.com/deeplearning/tensorrt/install-guide/index.html#installing)

When I run "dpkg -l | grep TensorRT", it shows libnvinfer7 libraries.
Please help, than you very much!

Seioch, 2020-07-21:
I had to upgrade cmake to 3.17.3 to get this library to be found. I think the repo has a min version of 3.15 for cmake.

minhtcai, 2020-07-22:
I'm using 3.18.0-rc4 though.

minhtcai, 2020-07-22:
Found the answer here! Need to set path here in cmake/findcudnn.cmake
#62

closed 2020-07-31 by minhtcai