ceccocats/tkDNN #90: How to get the inference label(center_x,center_y,H,w) like YOLOv4

wayne-blog, 2020-07-22:
I noticed the instruction in this program:
./demo <network-rt-file> <path-to-video> <kind-of-network> <number-of-classes> <n-batches> <show-flag>
and six parameters provided, but I didn't find how to get the result of inference.
As in official YOLO v4, I could do this:
To process a list of images data/train.txt and save results of detection to result.json file use: darknet.exe detector test cfg/coco.data cfg/yolov4.cfg yolov4.weights -ext_output -dont_show -out result.json < data/train.txt

To process a list of images data/train.txt and save results of detection to result.txt use:
darknet.exe detector test cfg/coco.data cfg/yolov4.cfg yolov4.weights -dont_show -ext_output < data/train.txt > result.txt

Could you help me about this issue, thanks a lot!

mive93, 2020-07-22:
Hi, you may want to take a look at the map demo instead: https://github.com/ceccocats/tkDNN#map-demo.
It will run the selected network on a list of images and will create a json file with the detections.

biparnakroy, 2020-07-25:
@wayne-blog  couple of days back i did have a project where i need tkdnn to produce YOLO like detection labels, so i made few changes in the code base which gave the desired results, here is my fork: https://github.com/biparnakroy/tkDNN, when you run demo you will get the yolo labels printed on your terminal. I had also added a few darknet like features like http stream.
ps: the use case that we both have is highly specific and that's the only reason i refereed you to my fork

wayne-blog, 2020-07-26:
> Hi, you may want to take a look at the map demo instead: https://github.com/ceccocats/tkDNN#map-demo.
> It will run the selected network on a list of images and will create a json file with the detections.

Thank you, I didn't get the point how to make it after I read that part.

wayne-blog, 2020-07-26:
> @wayne-blog couple of days back i did have a project where i need tkdnn to produce YOLO like detection labels, so i made few changes in the code base which gave the desired results, here is my fork: [...]

Thank you so much, really help me a lot, I'll check it out!

mive93, 2020-07-29:
@wayne-blog which part?
Have you succeed? Do you need further details?

mive93, 2020-09-11:
Closing for inactivity, 
feel free to reopen.

closed 2020-09-11 by mive93