ceccocats/tkDNN #91: jetson tx2  use yolov4 tensorrt  with demo/yolo_test.mp4,has only 7 fps, is that ok??

jcyhcs, 2020-07-22:
jetson tx2 , use tkDNN, jetpack4.2  TensorRT5.0.6, LOOK THE PICTURE:
why slow?

[image]

mive93, 2020-07-22:
Hi @jcyhcs 
It's completely normal, have you checked the results here: https://github.com/ceccocats/tkDNN#results?
If you want to achieve more FPS you can 
- active jetson_clock ( ```sudo jetson_clock``` )
- use FP16 instead of FP32
- use batching

Everything is explained in the README.

jcyhcs, 2020-07-22:
ok，this picture ， is that mean tx2 fp32 batch 1  is  7.30 fps？ i suggest  you should use " . "point, not use "," comma.
my tx2 already MAXN mode.
my result almost nearly yours. 

[image]

closed 2020-09-11 by mive93