ceccocats/tkDNN #92: about yolov4 with channel pruning

Hipeppa, 2020-07-24:
when i use yolo4 with channel pruning,it always with question
[image]
[image]

Hipeppa, 2020-07-24:
if i use the cfg produced by the pruned yolov4,it has this problem, if i use the yolo4.cfg, but i modify yolo4.cfg's channels according to the pruned yolo4,it can function properly.

ceccocats, 2020-07-24:
It can be anche issue of the parser, can you share the exact cfg you are using? Have you tried to run "diff" on the 2 files?

Hipeppa, 2020-07-24:
[1.txt](https://github.com/ceccocats/tkDNN/files/4970326/1.txt)
[2.txt](https://github.com/ceccocats/tkDNN/files/4970327/2.txt)
the 1.txt that can have problem like the picture, the 2.txt can use normally, but the 2.txt is modify by the 1.txt

ceccocats, 2020-07-27:
We do not support spaces in the parser
<key>=<value>  is ok
<key> = <value>  is not ok

Hipeppa, 2020-07-27:
thank u very much，i know .

MichaelCong, 2020-08-04:
> [1.txt](https://github.com/ceccocats/tkDNN/files/4970326/1.txt)
> [2.txt](https://github.com/ceccocats/tkDNN/files/4970327/2.txt)
> the 1.txt that can have problem like the picture, the 2.txt can use normally, but the 2.txt is modify by the 1.txt

what project do you used for channel pruning?

mive93, 2021-01-04:
Closing for now, feel free to reopen.

closed 2021-01-04 by mive93

Hipeppa, 2021-01-05:
> > [1.txt](https://github.com/ceccocats/tkDNN/files/4970326/1.txt)
> > [2.txt](https://github.com/ceccocats/tkDNN/files/4970327/2.txt)
> > the 1.txt that can have problem like the picture, the 2.txt can use normally, but the 2.txt is modify by the 1.txt
> 
> what project do you used for channel pruning?

https://github.com/tanluren/yolov3-channel-and-layer-pruning.git