ceccocats/tkDNN #95: mAP in tkDNN FP32 mode lower than FP16 mode and Darknet

huyhung411991, 2020-07-28:
Hi,
I trained different models of Yolov3-tiny and Yolov3-tiny-3l using AlexeyAB Darknet, used tkDNN to test the mAP in both FP32 and FP16 modes with the parsed weights on customized dataset. 
However, in general the mAP in FP32 mode is much lower than I tested in Darknet, and even slightly lower than in FP16, which should be impossible?
Please help.

Model | Non-TRT | TRT FP32 | TRT FP16
--- | ---
Yolov3-tiny | 512 | 73.9 | 64.7819 | 64.7819
Yolov3-tiny-3l | 512 | 81.9 | 72.455 | 72.5486
--- | ---
Yolov3-tiny | 768 | 80.5 | 74.5484 | 74.455
Yolov3-tiny-3l | 768 & 87.6 | 80.1072 | 80.0963
--- | ---
Yolov3-tiny | 1024 | 85.6 | 78.7161 | 78.7045
Yolov3-tiny-3l | 1024 | 89.1 | 81.3164 | 81.2897

mive93, 2020-07-29:
Hi @huyhung411991 
how did you compute the map? 
- with which software
- with which confidence threshold
- on which dataset?

huyhung411991, 2020-07-29:
My Jetson TX2 currently running JetPack 4.3 with Ubuntu 18.04, CUDA 10.0, cuDNN 7.6.3 and TensorRT 6.0.1.

I tested on a 1000 images of KITTI dataset, with map_points = 101, IoU_thresh = 0.5, conf_thresh = 0.25.

The input sizes for training and testing the models follow the aspect ratios of the original images, for example: 512x160, 640x192, 768x224.

huyhung411991, 2020-07-29:
I created the rt files and tested with an older commit of tkDNN, also retested with latest commit, but the results have the same issue.

lzyang2000, 2020-08-01:
I can also confirm it happening with YOLOv4:
Tested on visdrone dataset, 512x512, map_points = 0, IoU_thresh = 0.5, conf_thresh = 0.25.

darknet original: MAP@0.5 IOU: 37.1%, avg precision: 0.62, avg recall: 0.59
darknet tensorRT FP32: MAP@0.5 IOU: 29.7%, avg precision: 0.21, avg recall: 0.37
darknet tensorRT FP16: MAP@0.5 IOU: 29.7%, avg precision: 0.21, avg recall: 0.37

I used a Xavier AGX with JetPack 4.4, CUDA 10.2, TensorRT 7.1.3, cuDNN 8.0.

mive93, 2020-08-01:
Sorry for the late reply, I didn't see the answer.
The problem is the confidence threshold. You both have to change it in DetectionNN.h. There the conf is 0.3. Try using 0.05 instead and you will get the correct values.

I know, it's hardcoded :/ I have changed it in a branch, but I still have to pull it on the master. 
And yes, FP32 is better than FP16 (but there is no big loss actually) and you don't have to set anything in the demo, just saying which rt you are using.

lzyang2000, 2020-08-01:
Thanks for the response! Yes I saw the threshold issue and changed the confidence threshold in detectionNN.h to 0.05, and in the demo/config.yaml I set confidence threshold to be 0.25, and got the values that I quote here:

> I can also confirm it happening with YOLOv4:
> Tested on visdrone dataset, 512x512, map_points = 0, IoU_thresh = 0.5, conf_thresh = 0.25.
>
> darknet original: MAP@0.5 IOU: 37.1%, avg precision: 0.62, [...]

If I change the confidence threshold to 0.05 in config.yml, I get 
darknet tensorRT FP32: MAP@0.5 IOU: 33.5%, avg precision: 0.26, avg recall: 0.47
darknet tensorRT FP16: MAP@0.5 IOU: 33.5%, avg precision: 0.26, avg recall: 0.47

I did round the values a bit, but still the performance drop when converting to tensorRT seems strange.

lzyang2000, 2020-08-01:
One thing I find strange - the visdrone dataset contains 12 labels, namely these:
```
ignored regions
pedestrian
people
bicycle
car
van
truck
tricycle
awning-tricycle
bus
motor
others
```
There are 12 classes in total, but if in config.yaml I change it to 11 classes and set the confidence threshold to 0.05, I can get 36.5% which is somewhat comparable.

My suspicion is that the ignored region class, being the 0th index, affects the calculation of mAP somehow.

lzyang2000, 2020-08-01:
So I finished testing the pretrained MSCOCO yolov4 weights, and also see a 8.7% decrease from 70% to 61.3% from mAP@0.5IOU.

mive93, 2020-08-06:
Hi @lzyang2000 

yes, there was a bit loss in tkDNN. I have fixed it. 
See #103

lzyang2000, 2020-08-06:
Yes I also figured it out yesterday looking at Darknet's code - turns out they are evaluating the mAP at confidence threshold = 0.005, so basically the confidence threshold in DetectionNN.h (I did not pull) and config.yaml need to be at least 0.005.

Now it works. 🤣

closed 2020-09-11 by mive93