ceccocats/tkDNN #97: Can't find .trt file

maohu-96, 2020-07-28:
Hi！ @ceccocats @mive93 
Sorry to disturb you!
I created the build file under tkdnn and copied my cfg and weight files, but the .trt file was not found after make. Is my operation wrong?
[image]

mive93, 2020-07-29:
Hi @maohu-96, have you followed the README? 
You should run a test to create a trt model. 
If you're using a model that is not in this repo, you need to first export the weights (https://github.com/ceccocats/tkDNN#1export-weights-from-darknet).
Then you need to create a test similar to this one (https://github.com/ceccocats/tkDNN/blob/master/tests/darknet/yolo4.cpp) and in case, change line 12-19 according to your model.
Finally, you will be able to run the demo, once you have run your test (https://github.com/ceccocats/tkDNN#run-the-demo).

maohu-96, 2020-07-29:
> Hi @maohu-96, have you followed the README?
> You should run a test to create a trt model.
> If you're using a model that is not in this repo, you need to first export the weights [...]

Thank you very much for your reply! @mive93 
Sorry to disturb you. I exported my model and copied my cfg and names file. An error occurred after make. How can I solve it?
[image]

mive93, 2020-09-11:
Continued in #98 
Closing for now,
feel free to reopen.

closed 2020-09-11 by mive93