ceccocats/tkDNN #98: libtkDNN.so error : Undefined reference

maohu-96, 2020-07-29:
Hi!
The following error appears after make, what should I do, thank you!
[image]

mive93, 2020-07-29:
Hi @maohu-96 ,

have you installed yaml-cpp? Which version? 
This the min required version needed: yaml-cpp 0.5.2 (sudo apt install libyaml-cpp-dev)

maohu-96, 2020-07-29:
> Hi @maohu-96 ,
> 
> have you installed yaml-cpp? Which version?
> This the min required version needed: yaml-cpp 0.5.2 (sudo apt install libyaml-cpp-dev)

Hi @mive93 
Thanks for your reply, I installed yaml-cpp (0.5.2)

praveenbadimala, 2020-09-04:
libyaml incompatibility with libboost version fix:

a) download libyaml-cpp dev 0.5.2.0 from github
b) change the below import statements wherever it occurs:

#include <boost/utility.hpp>

to

#if BOOST_VERSION >= 106700
#include <boost/next_prior.hpp>
#else
#include <boost/utility.hpp>
#endif

compile the library with the below flags:
cmake and then make and later make install
env CFLAGS='-fPIC' CXXFLAGS='-fPIC' cmake -DCMAKE_BUILD_TYPE=Release ..

then tkdnn compilation should go fine without issue:

Issue references:
https://github.com/jbeder/yaml-cpp/issues/682
https://github.com/arvidn/libtorrent/pull/2929/commits/253ae7ebeaa8287877b7fad2d8b2c083e445340c

mive93, 2020-09-11:
Closing for now,
feel free to reopen.

closed 2020-09-11 by mive93

Goru1890, 2020-11-25:
I'm having the same problems as @maohu-96 and I installed the yaml-cpp version 0.5.2-5ubuntu1.
I don't really understand what @praveenbadimala explained above.

Goru1890, 2020-11-25:
My error is actually a little bit different

[image]