ceccocats/tkDNN #101: Inference on a single frame much slower than mp4?

Seioch, 2020-07-30:
Hi Authors, 

I wrote a simple program using tkDNN to do inference on a single frame. Before running this program, I created a new TensorRT file with FP16 enabled using a yolov4_416 configuration. 
```

int main(int argc, char* argv[]){
    if(argc < 2){
        FatalError("Not enough parameters, must provide path to yolov4 rt file and image");
    }
    tk::dnn::Yolo3Detection yolo;
    tk::dnn::DetectionNN* network = &yolo;

    network->init(argv[1], 38, 1);
    std::cout << "Yolo initialized" << std::endl;
    
    cv::Mat image;
    std::vector<cv::Mat> dnnInput;
    image = cv::imread(argv[2], cv::IMREAD_COLOR);   // Read the file

    if(! image.data )                              // Check for invalid input
    {
        FatalError("Image not valid")
    }

    dnnInput.push_back(image);  // Add the image to a vector of frames 

    network->update(dnnInput, 1); // do the inference 

    std::vector<std::vector<tk::dnn::box>> boxes = network->getRawBoundingBoxes();

    std::cout << "Number of BBs: " << boxes[0].size() << std::endl;
    
    for (int i = 0; i < boxes[0].size(); i++) {
        std::cout << "Box " << i << " contents: \n";
        boxes[0][i].print();
    }

    std::cout << "Inferred in " << *network->stats.begin() << " milliseconds\n";
}
```
When I run the demo program with FP16 inference enabled, it reported an average of 10.9 ms per frame, with a max frame inference time of 55ms. However, inferring with my test program reports a 25ms inference time per frame when run repeatedly. Is there a reason for a difference in inference speed?

Seioch, 2020-08-03:
Update: I did some performance analysis on inference in a single video and I was able to produce  this result: 

[image]

It seems that the first update() call takes a significant amount of time, and then the rest of the inference seems to be fast. I don't know why this is the case, however it may explain why inferring on one frame seems so slow.

mive93, 2020-08-06:
Hi @Seioch 

sorry for the late reply.
Yes, it is totally normal. The first iteration is always the worst case. This is due to the fact that the rt model need to be loaded on the GPU. Moreover after the first iteration, cache can be exploited.
It is something very usual, not only with out sotfware.

closed 2020-09-11 by mive93