ceccocats/tkDNN #102: tensorrt + yoloV4 run more slowly than darknet

basailuonaaa, 2020-08-03:
On the RTX 2080Ti (CUDA 10.0, TensorRT 6.05, Cudnn 7.6.01):
for  yoloV4 608x608, source image is 1080*1920:

darknet source code process each image about 40 milliseconds, however, tensorrt + yoloV4  process each image about 45 milliseconds.

What is the reason, how should I solve it?

mive93, 2020-08-03:
Hi @basailuonaaa,

It is weird. I have performed the same test, on the same GPU, on a video with the same resolution at FP32 precision.
Preprocessing + inference + postptocessing takes ~18 ms.
Including also all the other operations performed in a loop cycle, it takes ~30ms.

mive93, 2020-09-11:
Closing for inactivity, 
feel free to reopen.

closed 2020-09-11 by mive93