ceccocats/tkDNN #109: this is not yolo3

Nightaway, 2020-08-13:
I exported a yolo2 model using https://git.hipert.unimore.it/fgatti/darknet.git. Then i writed a cpp program to export runtime file. Everything looks work. But it say "this is not yolo3" .

detection
mytest_fp32.rt
New NetworkRT (TensorRT v7.13)
Float16 support: 1
Int8 support: 1
DLAs: 2
create execution context
Input/outputs numbers: 2
input idex = 0 -> output index = 1
Data dim: 1 3 416 416 1
Data dim: 1 125 13 13 1
RtBuffer 0   dim: Data dim: 1 3 416 416 1
RtBuffer 1   dim: Data dim: 1 125 13 13 1
this is not yolo3
/home/nx/Desktop/tkDNN/src/Yolo3Detection.cpp:18
Aborting...

Nightaway, 2020-08-13:
#include<iostream>
#include<vector>
#include "tkdnn.h"
#include "test.h"
#include "DarknetParser.h"

int main() {
    std::string bin_path  = "mytest";
    std::vector<std::string> input_bins = { 
        bin_path + "/layers/input.bin"
    };
    std::vector<std::string> output_bins = {
        bin_path + "/layers/output.bin",
    };
    std::string wgs_path  = bin_path + "/layers";
    std::string cfg_path  = "mytest/yolo-voc.cfg";
    std::string name_path = "mytest/voc.names";

    // parse darknet network
    tk::dnn::Network *net = tk::dnn::darknetParser(cfg_path, wgs_path, name_path);
    net->print();

    //convert network to tensorRT
    tk::dnn::NetworkRT *netRT = new tk::dnn::NetworkRT(net, net->getNetworkRTName(bin_path.c_str()));
    
    int ret = testInference(input_bins, output_bins, net, netRT);
    net->releaseLayers();
    delete net;
    delete netRT;
    return ret;
}

ceccocats, 2020-08-14:
The demo detector support only yolo3 and above.
This because the end layer of yolo2 is different from yolo3 and yolo4.

closed 2020-08-21 by Nightaway